melkati/CO2-Gadget #26: Add support for Anaire Cloud - Working with test.mosquitto.org:1883 fails with mqtt.anaire.org:80

melkati, 2021-11-21:
Works fine when connecting to broker test.mosquitto.org:1883 but fails on connection to broker mqtt.anaire.org:80

melkati, 2021-12-08:
@anaireorg
@pamonmo
@alogracia

¿Me echáis una manita con esto, porfa?

He vuelto a hacer pruebas y no consigo conectar con el broker. 

Ni siquiera lo consigo desde Nodered.

Me surge una duda... ¿Se hace alguna comprobación sobre el DeviceId? ¿Puede ser inventado o debe construirse como lo hacéis vosotros? En Nodered es uno inventado pero en CO2 Gadget he copiado vuestra función para calcularlo, por lo que no debería se problema.

https://github.com/melkati/CO2-Gadget/blob/f8c5f85041bce7c5d2e8223838dfdc9bc595cf19/CO2_Gadget_Anaire-Cloud.h#L314-L324

acmatl, 2021-12-08:
Buenas noches, no me entero con el código pero igual te puede servir esto.
https://github.com/anaireorg/anaire-cloud/blob/main/stack/mqttforward.yaml
Creo que en este fichero construyen un device_id para mqtt, que no sé si es lo mismo.
Saludos

melkati, 2021-12-09:
> Buenas noches, no me entero con el código pero igual te puede servir esto. https://github.com/anaireorg/anaire-cloud/blob/main/stack/mqttforward.yaml Creo que en este fichero construyen un device_id para mqtt, que no sé si es lo mismo. Saludos

Muchas gracias, Carlos.

Lo estado viendo pero me parece que eso no explica por qué el cliente no se conecta con el broker.
Probablemente tengan que ayudarnos los chicos de @anaireorg con esto.

pamonmo, 2021-12-09:
Buenas,

Perdonad que no hemos podido ponernos con esto aún, estamos con unas actividades para Fundación Telefónica y con algún tema personal que nos han tenido más ausentes de lo que nos gustaría. No se hace comprobación del id, con lo que eso no debería ser.

Mario, dinos cuál es el id y cuando me pueda meter a mirarlo reviso logs a ver si dice algo de tu dispositivo, y si no montamos una sesión de troubleshooting.

Saludos,
Pablo

Enviado el: jueves, 9 de diciembre de 2021 11:50
Para: melkati/CO2-Gadget ***@***.***>
CC: PABLO MONTES MORENO ***@***.***>; Mention ***@***.***>

Este mensaje y sus adjuntos se dirigen exclusivamente a su destinatario, puede contener información privilegiada o confidencial y es para uso exclusivo de la persona o entidad de destino. Si no es usted. el destinatario indicado, queda notificado de que la lectura, utilización, divulgación y/o copia sin autorización puede estar prohibida en virtud de la legislación vigente. Si ha recibido este mensaje por error, le rogamos que nos lo comunique inmediatamente por esta misma vía y proceda a su destrucción.

The information contained in this transmission is privileged and confidential information intended only for the use of the individual or entity named above. If the reader of this message is not the intended recipient, you are hereby notified that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this transmission in error, do not read it. Please immediately reply to the sender that you have received this communication in error and then delete it.

Esta mensagem e seus anexos se dirigem exclusivamente ao seu destinatário, pode conter informação privilegiada ou confidencial e é para uso exclusivo da pessoa ou entidade de destino. Se não é vossa senhoria o destinatário indicado, fica notificado de que a leitura, utilização, divulgação e/ou cópia sem autorização pode estar proibida em virtude da legislação vigente. Se recebeu esta mensagem por erro, rogamos-lhe que nos o comunique imediatamente por esta mesma via e proceda a sua destruição

melkati, 2021-12-09:
Hola Pablo.

Estupendo. Muchas gracias!

Acabo de actualizar el branch anaire-cloud porque estaba muy alejado del master, he grabado un medidor con ese branch y se ha intentado conectar a mqtt.anaire.org:80 a las 19:17 con DeviceId d26c4

```
19:17:51.301 > ESP32 Chip model = ESP32-D0WDQ6 Rev 3.   This chip has 2 cores and 4MB Flash.
19:17:51.301 > Anaire Device ID: d26c4
19:17:51.301 > Attempting to connect to the Anaire Cloud MQTT broker mqtt.anaire.org:80
19:18:06.548 > Attempting Anaire Cloud MQTT connection...failed, rc=-4 try again in 5 seconds
```
Si miras en el log esa hora debería estar.

Unos minutos más tarde probé con la IP del broker, por si acaso había algún problema de resolución de nombres, con el mismo resultado.

19:36:17.825 > Anaire Device ID: d26c4
19:36:17.825 > Attempting to connect to the Anaire Cloud MQTT broker 18.203.38.54:80
19:36:32.974 > Attempting Anaire Cloud MQTT connection...failed, rc=-4 try again in 5 seconds

El código que estoy utilizando es prácticamente el vuestro [anaire-cloud/CO2_Gadget_Anaire-Cloud.h](https://github.com/melkati/CO2-Gadget/blob/anaire-cloud/CO2_Gadget_Anaire-Cloud.h) con las modificaciones mínimas para que funcione en CO2 Gadget.

Entras las 19:30 y las 19:35 hice varias pruebas de conexión desde Nodered, con varias configuraciones, con el ClientId 234349

Las pruebas se ha hecho desde la IP 79.144.202.159

Muchas gracias por tu ayuda. :-)

melkati, 2024-01-07:
After two years, with limited interest or assistance, I have decided to close this. I will reopen it in the future if there is renewed interest.

closed 2024-01-07 by melkati