melkati/CO2-Gadget #35: Add LoRa connexion.

roberbike, 2021-12-08:
Add LoRa protocol to connect sensors over long distances and send data if no wifi network is available.

melkati, 2021-12-08:
Very good idea! Thank you for your suggestion! 😀

In fact I already bought some LoRa modules to start testing.

As soon as ESP-Now is working I'll start working on LoRa.

I guess ESP-Now will not be very difficult as I already have the communications code for ESP-Now write in the "Ultra Low Power Co2 Monitor" (unpublished by now, sorry). Left... to add to preferences, configuration menu, etc...

freddujour, 2022-01-10:
Hi, If you need some testers, I can support with the lora heltec development board and rfm95 chipset

melkati, 2022-01-11:
Guys, I would like to know more from you about this feature request.

AFAIK (I'm not a LoRa expert, just had some play with TinyGS project) there may be two "levels" to implement it:

1. A basic level where CO2 Gadget only has to send the data via LoRa and "some receiver" (probably the one I already have on development for the ESP Now implementation) receives the data, decodes the payload format and re-sends it some other way (MQTT, for example, as the ESP Now Gateway is doing right now). **Note that this level is not LoRaWAN, just LoRa.**
2. An "advanced level" where CO2 Gadget can be integrated in a LoRaWAN network (as The Things Networks or another) so it must talk the official LoRaWAN protocol. Much more complex to do than basic level.

In order to better understand this, I need some more information.

- What are your use cases?
- Do you need real LoRaWAN implemented or just the basic LoRa.
- If LoraWAN, do you have a **real LoRaWAN Gateway** to test the implementation and are willing to invest some time on development (as I don't have a LoRaWAN Gateway myself)?

Many thanks in advance.

roberbike, 2022-01-12:
I have 2 ttgo Lora and 1 ttgo t-beam. My target will be make 1 mono channel gateway connect to chirpstack, or TTN V3, and other Lora device with co2 gadget. Lorawan is needed for the gateway, this is the easy part. The co2 gadget only need to be Lora, I think, gateway is the device to receipt the datas and resent to the iot net.

melkati, 2022-01-12:
I will, probably, do some testing in the next few days.

After some help received in the [Spanish LoRa / LoRaWAN Telegram group](https://t.me/LoRa_LoRaWAN/26688), looks like the way to go si to **support full LoRaWAN** and use the [mcci-catena/arduino-lmic](https://github.com/mcci-catena/arduino-lmic) library.

I already have some different LoRa modules (raw modules and modems) and boards (the LoRa 32 from Heltec) so I think I'm ready for testing.

I'm a little worried about ESP32 memory use and if it will fit with the rest of features (as WiFi, BLE and ESP Now) on CO2 Gadget (as WiFi, BLE and ESP Now). Maybe a different compiled flavor is necessary to exclude some features. Testing will say...

freddujour, 2022-01-12:
Hi,

I fully support the implementation of lorawan. I have my private lorawan
stack with mikrotik router+chirpstack server and a few devices using
arduino clone atmega32U4 +rfm95 or heltec esp32 lora. I think if you
implement lorawan, wifi is not making sense anymore. For testing purposes,
I can use some ttgo lora as well.

melkati, 2024-01-07:
After two years, with limited interest or assistance, I have decided to close this. I will reopen it in the future if there is renewed interest.

melkati, 2024-06-05:
He construido e instalado en casa un Gateway LoraWAN conectado a TTN con la única intención de incluir esta funcionalidad en CO2 Gadget (bueno, y de paso contribuir a incrementar la cobertura de TTN en mi ciudad).

Retomará la inclusión de esta funcionalidad en cuanto tenga algo de tiempo.