melkati/CO2-Gadget #36: Auto detect SSID WIFI access points

Coscolin, 2021-12-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

freddujour, 2022-02-16:
Would be great also if in an instable wifi environment, the gadget would try the reconnect  to the AP in case the link is lost (including the MQTT broker). At this moment both are going to off.

melkati, 2022-02-17:
> Would be great also if in an instable wifi environment, the gadget would try the reconnect to the AP in case the link is lost (including the MQTT broker). At this moment both are going to off.

Hi @freddujour 

Thanks for your suggestion.

I can do it but, please, open your own feature request so it doesn't get mixed with the one to list available SSID.

freddujour, 2022-02-17:
Sorry, done. You can delete my comment.
Thanks for the great job

melkati, 2022-02-17:
Thank you. Follows at #89

closed 2024-01-08 by melkati