melkati/CO2-Gadget #39: Change the sensor's co2 measurement time interval to optimize battery consumption

Coscolin, 2021-12-08:


melkati, 2021-12-09:
Hello Sergio and thank you for this interesting feature request submission.

It's very **doable and desirable** to have but first we have to short out some questions regarding this functionality in order to implement it the right way. There are some points that worries me:

1. The fact that each time the user changes the measurement time interval implies that **the sensor has to be calibrated again**. I'm not sure if the user will be aware of this and will calibrate the sensor or just leave it as is, receiving bad readings.
2. This feature is dependant on the sensor used (remember that CO2 Gadget supports a few sensors). As far as I'm aware, the only sensor supporting this functionality is the Sensirion SCD30 (and the ultra low power sensors as Senseair S11 Sunrise,  Sensirion SCD41 and Cubic CM1106SL-NS).
3. The added complexity for the user and maybe lack of usability.
4. This is not a low power CO2 meter (as you know I've another project for that) and the improvement in autonomy will not be very big.
5. Will the ratio effort/benefit and the number of users using it be good enough to be worth? 

Regarding **point 1**, the are some ideas that come to mind to solve it:
- Invalidate the calibration and make sure the meter doesn't shows CO2 values again until the sensor is recalíbrated. The issue is that most users don't know how to do a correct calibration and will end with wrong readings. Also, it implies to have to implement this program logic, taking care of this flag and making the meter working in a different way (inclusive between reboots) 

In regard to **point 2**, the ultra low power sensors as Senseair S11 Sunrise,  Sensirion SCD41 and Cubic CM1106SL-NS, supports some thing similar to this and I would like them to be supported but they need deep changes in the sensor management logic. Don't know if it's better to include this functionalities when including these sensors.
The actual sensors library (CanAirIO Sensors Lib) doesn't support ultra low power sensors and there are two options: 
- Contribute to the library with the modifications needed to support these sensors (big refactoring needed)
- Utilize different, independent, libraries for this sensors.

Anyway, I want to implement this, one way or another, so I add this feature request to the Roadmap.

Coscolin, 2021-12-10:
Hello Mario,

I think this project is designed initially with SCD30, as you recommend in your blog, so although it can be only available in this sensor, is not a problem.
Only shows this option in menu when SCD30 (or SCD4x if you want) is detected.

I think is better to contribute to "CanAirIO Sensors Lib" to support it than use independent libraries.

About point 1: maybe user must be warned about it before change time interval, and automatically re-calibrate sensor after changes.

Point 4: if user uses a eInk board, this enhacement it will be very relevant to low power consumption. I don't know how much reduce with TFT display, but I think it's a good improvement.

Thanks.

melkati, 2021-12-20:
While we think of a better way of doing it (and see what happens with the ultra low power sensors) I come with a way to change measurement period **for advanced users** (as I want to avoid people playing with this value without knowing what are they are doing).

I want to keep it a somewhat hidden and undocumented feature, for the moment (not a secret, a it's described right here in a public message and it's clearly visible on the source code).

The way to change the measurement interval is via the web interface with the following URL:

`<CO2-GADGET_IP>/update?MeasurementInterval=seconds`

For example to set the reading interval to 15 seconds in a CO2 Gadget with IP `192.168.1.222`, the URL to access will be:

`http://192.168.1.222/update?MeasurementInterval=30`

**WARNING: Re-calibrate your sensor each time you change the measurement interval**, otherwise readings will be wrong, I will double check with witch sensors a re-calibration is a must and update this message (for sure with the SCD30, SCD41 and SCD41).

closed 2021-12-20 by melkati

melkati, 2021-12-21:
Soon a new version sill change the URL from 

`<CO2-GADGET_IP>/update?MeasurementInterval=seconds`

into:

`<CO2-GADGET_IP>/settings?MeasurementInterval=seconds`

This is to free the /update url for Over The Air updates.