melkati/CO2-Gadget #42: Secure WiFi and MQTT passwords

Coscolin, 2021-12-10:
First at all, hide passwords in menu.

If you want to change any password you must to type it again completely.

Maybe is possible to store passwords with MD5 encrypt / decrypt code?

melkati, 2021-12-10:
> First at all, hide passwords in menu.
> 
> If you want to change any password you must to type it again completely.
> 
> Maybe is possible to store passwords with MD5 encrypt / decrypt code?

There are two feature requests in this issue:

1. There is already an entry in roadmap to [hide passwords in menu](https://github.com/melkati/CO2-Gadget/projects/2#card-73265732).

2. MD5 is a one way trip. You can store a password's MD5, and then check, for example, if the user types the same password if it's the same password you calculated the MD5 last time, or not, What isn't possible is to convert the MD5 back into the original password (needed to connect to WiFi or to the MQTT broker),

I will think about some solution to store encrypted passwords but it's not an easy task. Will have to study documentation about  ESP32 Secure Boot, Flash Encryption, etc.

melkati, 2021-12-12:
Hide passwords in menu implemented in last commit to master
WIFI_PRIVACY defined by default (hides WiFi and MQTT passwords in menu and logs)

closed 2021-12-12 by melkati