melkati/CO2-Gadget #43: Add and test CO2 sensors others than SCD30 and SCD4x (already tested)

melkati, 2021-12-12:
# TODO

- [x] Include in menu the selection for CO2 sensors others than SCD30 and SCD4x
- [x] Save sensor selection to preferences (NVR)
- [x] Include the necessary logic lo support sensor selection (including serial connected sensors)
- [x] Test SCD30 Sensor
- [x] Test SCD41 Sensor
- [ ] Test MHZ19 Sensor
- [ ] Test CM1106 Sensor
- [ ] Test SENSEAIRS8 Sensor

freddujour, 2021-12-12:
Hi,

I have successfully tested the MHZ19 sensor. Found out how to proceed by modifying CO2_Gadget_Sensors.h
Changing line 74 with : _sensors.init(sensors.Mhz19);_

Keep going, it is a very nice project.

melkati, 2021-12-12:
Hi Frédéric.

Thank you very much. :-)

You don't need to modify the source code to select different sensors.

Just:

- Compile and flash with code in [Feature-Select-CO2-Sensor branch](https://github.com/melkati/CO2-Gadget/tree/Feature-Select-CO2-Sensor)

- Go to menu **Configuration -> CO2 Sensor** and select option "**CO2 Sensor**" now you can select the sensor with "Enter" (long press upper button on TTGO T-Display of from the serial console with key "*"). 

After each "Enter" give CO2 Gadget some time to try to init the sensor (one-two seconds), you will notice the menu unresponsive during this time. 
This is something I want to improve before releasing this feature.

[image]

Done!

I'll merge this feature in master branch as soon as make some improvements (as the delay in menu) and do some more tests.

You are more than welcome to test this branch as I already tested this feature with Senseair S8 and Cubic CM1106 but not with MH-Z19. Thank you!

melkati, 2021-12-15:
Released as v0.5.001

- Add support for serial CO2 Sensors as MH-Z19(A/B/C/D), CM1106, Senseair S8
- Reworked menu system to make it easier to select sensors and avoid delays while selection
- WiFi privacy defined by default (it doesn't show passwords in menu or serial terminal)
- Lots of fixes and improvements

@freddujour

closed 2021-12-15 by melkati