melkati/CO2-Gadget #74: Changes

ljrnjh1964, 2022-01-02:
Hello. Please output the СO2 signals to separate pins for green, orange and red.

melkati, 2022-01-02:
Hi

Please, can you develop your request a little more?

There are already individual pin outputs for green, orange and red. 

What do you want to accomplish?

ljrnjh1964, 2022-01-03:
Hello. I need separate pins for orange and red for separate connection of the relay, the contacts of which close the capacitor leads to control the fan speed. You have orange and red at 32 pins. 
[image]

melkati, 2022-01-03:
Hi,

Sorry, my mistake. It was an error on documentation.  Orange is on pin 32 and red in 33

https://github.com/melkati/CO2-Gadget/blob/d69294294bdaa790c71a2eee84285cf9e1bac1d1/platformio.ini#L60-L68

closed 2022-01-03 by melkati

ljrnjh1964, 2022-01-03:
Thanks a lot for the clarification.