melkati/CO2-Gadget #80: Turn on display when CO2 values rises limits

Coscolin, 2022-01-17:
During battery operation, if user has selected display automatic turn off after X seconds it could be neccesary turn on display if measures rise to high values.

I think it could be turn on when measures increase or decrease 50 or 100 ppm from last value displayed, and turn it off again after X seconds.

When measures are at upper high risk limit, it could be neccesary to show values at least once time every 10 minutes, although don't change values.

Security must be more than battery optimization.

Coscolin, 2024-02-01:
Using bufferLoop() we can include a call to function:
void IRAM_ATTR buttonUpISR() {
    if (actualDisplayBrightness == 0)  // Turn on the display only if it's OFF
    {
#if defined(SUPPORT_OLED) || defined(SUPPORT_TFT)
        setDisplayBrightness(DisplayBrightness);  // Turn on the display at DisplayBrightness brightness
#endif
        lastTimeButtonPressed = millis();
    }
}
in CO2_Gadget_Buttons.h to turn on display when CO2 values rises limits.
This call only change lastTimeButtonPressed value to turn on display if it's off

melkati, 2024-02-01:
Good idea.

We must think about it, as there are some things planned to implement (as display management) that can collide with the buttons.

Anyway, probably we should not call this function directly as it's an ISR (interrupt service routine) and should only be called by a hardware interrupt (but we can duplicate the functionality on another similar one, no problem).

Let's talk about it later, when display management functionality is done...

Also, it must be after finish #154 as it also needs buttons to stop buzzer beeping.

Probably we should think of a generic way to stop an "abnormality" by briefly pressing a button.