melkati/CO2-Gadget #82: Hide information on the display

juanluis8686, 2022-01-18:
In some cases, we do not have a temperature / humidity sensor or we do not want to see these values on the screen. It would be nice to be able to hide this information.

melkati, 2022-01-18:
Good feature request!
I'll implement it.
Thank you.

melkati, 2022-01-18:
It will add options to display configuration menu to show/hide (show by default) Temp, Humidity and Battery, I don't think CO2 is needed right now, I'll implement it when there is also PM 2.5 support.

- [ ] Add functionality to code and include in saved preferences
- [ ] Add menu selector to show/hide: Temp, Humidity and Battery
- [ ] Test functionality 
- [ ] Take configuration menu screenshot
- [ ] Create documentation on how to use

closed 2022-01-22 by melkati