melkati/CO2-Gadget #86: WIFI password length limit

zorg2015, 2022-01-29:
HI,
I have build a CO2 sensor using the CO2 Gadget from mariete. It works fine in its standalone configuration. Now I would like to activate the wifi. unfortunately, it seems that the length of the password is limited to 28 characters whereas my wifi password uses 37 cars.  Do you confirm this limit ? If yes is it possible to increase this length ?

Regards

melkati, 2022-01-30:
Hi.

Thank you for reporting this.

Yes, the actual max password length is around 30 characters. 

I will increase the length as to the maximum password length standard that allows a max length of 63 characters.

I hope to have this fix ready very soon.

Best regards.

melkati, 2022-01-31:
Sorry for the delay. I'm working on this but I'm facing an issue while modifying the menu.

I'm awaiting for an answer from the menu's library author.

#https://github.com/neu-rah/ArduinoMenu/issues/378

zorg2015, 2022-02-05:
Thank you for having taken into account my request. I didn't think it would be necessary to modify a library but it seems rather logical. So, I think it'll be simpler for me to shorten my password even if I have many wifi devices...
Have a nice day

melkati, 2022-02-08:
I'm sorry fixing the menu is not in my hands right now.
I'll revisit this in the future to give a solution.

closed 2022-02-08 by melkati