melkati/CO2-Gadget #89: wifi and mqtt broker reconnect attempts in case connection is lost

freddujour, 2022-02-17:
Would be great also if in an instable wifi environment, the gadget would try the reconnect to the AP in case the link is lost (including the MQTT broker). At this moment both are going to off.

melkati, 2022-02-17:
Hi @freddujour

Thanks for your feature request.

It's a very good one. I suffer it myself when CO2 Gadget temporarily loses connection.

I just added it to roadmap.

hapklaar, 2022-04-30:
Would love to have this as well! It seems I need to re-enable MQTT almost every day. Maybe an option to just leave it enabled even when there is no network connection?

melkati, 2024-01-08:
"I have just released a new version with a revised WiFi and MQTT connection and reconnection scheme. I implemented a much more robust mechanism to reconnect in case of disconnection, while also avoiding battery depletion.

Now, if the WiFi connection is interrupted, CO2 Gadget attempts to reconnect five times, and if it is not successful, it waits for 5 minutes before trying again.
Now, CO2 Gadget does not disable either WiFi or MQTT due to connection drops.
The MQTT connection is retried each time a new WiFi connection is established.
If the MQTT connection is interrupted (but the WiFi connection is maintained), it attempts to reconnect every 15 minutes.

This is a first implementation. I plan to enhance it with customizable parameters and the possibility to make the wait periods between connections longer as time passes; for example, for every 5 minutes of disconnection, the time for the next retry increases by two minutes (added to the initial five minutes, which are configurable)."

closed 2024-01-08 by melkati