melkati/CO2-Gadget #96: Add support for ESP32-S3 (Lilygo T-Display-S3)

hapklaar, 2022-10-14:
Would you consider adding support for ESP32-S3 of in particular the T-Display-S3, to be able to install this software on this new Liligo device with the much bigger screen?

disruptivepatternmaterial, 2022-11-01:
If someone wants to help me with the pins in the platformio.ini flags to pass in, I have this board and have been trying to get it to work, my code runs fine, but I cannot get the display to come on...I cannot figure out if it is using those parameters or the User_Setup.h method that the TRT lib prefers.

From the board examples

/*ESP32S3*/
#define PIN_LCD_BL                   38

#define PIN_LCD_D0                   39
#define PIN_LCD_D1                   40
#define PIN_LCD_D2                   41
#define PIN_LCD_D3                   42
#define PIN_LCD_D4                   45
#define PIN_LCD_D5                   46
#define PIN_LCD_D6                   47
#define PIN_LCD_D7                   48

#define PIN_POWER_ON                 15

#define PIN_LCD_RES                  5
#define PIN_LCD_CS                   6
#define PIN_LCD_DC                   7
#define PIN_LCD_WR                   8
#define PIN_LCD_RD                   9

#define PIN_BUTTON_1                 0
#define PIN_BUTTON_2                 14
#define PIN_BAT_VOLT                 4

#define PIN_IIC_SCL                  17
#define PIN_IIC_SDA                  18

#define PIN_TOUCH_INT                16
#define PIN_TOUCH_RES                21

hapklaar, 2023-12-13:
Did you have any success in getting it to work on a T-Display-S3?

melkati, 2024-01-07:
I have just ordered a T-Display-S3 to try adding support for this board. My plan is to pick up the work that disruptivepatternmaterial did on this (and other contributions, thank you @disruptivepatternmaterial!) and continue with it.
If I can't make it work, I will start from scratch.

melkati, 2024-01-21:
There is a new version[ in the development branch ](https://github.com/melkati/CO2-Gadget/tree/development)with support for Lilygo t-Display S3, if you want to try it...

⚡️ Beta version of CO2 Gadget firmware v0.8.034-development

Help! 🙏

As you know, in these recent weeks, I've been working a lot on CO2 Gadget (at my own pace, of course), and I have a beta version ready with a lot of significant changes.

I need your help to test it and identify any bugs, which I'm sure there will be, before releasing it as a new version.

There are profound changes that affect the basic structure and logic of CO2 Gadget, so the more people who try it, the better.

✔️ Support for the ESP32 S3 microcontroller

✔️ Full support for the Lilygo T-Display S3 board with a larger screen and a resolution of 320x170

✔️ WiFi configuration from the mobile device via Bluetooth

✔️ Improvements in WiFi connection and reconnection

✔️ WiFi active by default

✔️ Send data via MQTT immediately upon connection without waiting for the first interval to complete

I appreciate it if you can test it:

✅ Anyone who is already using CO2 Gadget

✅ Those who have a T-Display S3 board and can connect a CO2 sensor

Thank you very much! 😊

hapklaar, 2024-01-21:
Very nice, it works!

- Could you include an option in the display config to turn on the backlight when running on battery? For this module this is off by default and can be enabled by [setting GPIO15 high](https://community.lilygo.cc/topic/711/unable-to-make-t-display-s3-work-with-lipo-battery).
- The small font doesn't look very good on this resolution, jagged/aliased. Could benefit from a different font.
- Large font could be a bit bigger to use more of the screen, maybe also other type with anti-aliasing?

Will test some more the coming days.
[image]

melkati, 2024-01-23:
> Very nice, it works!

Thank you! :-)

> * Could you include an option in the display config to turn on the backlight when running on battery? For this module this is off by default and can be enabled by [setting GPIO15 high](https://community.lilygo.cc/topic/711/unable-to-make-t-display-s3-work-with-lipo-battery).

Already done in the versión you tried. GPIO15 is high all the time.

There are some issues with display brightness  #125. I will come back to this issue in a few days.

You should configure the "Time to Off" to 0 to disable turning off the display when on battery (this is how it was working before the bug, I hope to fix it).

You can find all the options at: https://emariete.com/en/co2-meter-gadget/#Display_Config

[image]

> * The small font doesn't look very good on this resolution, jagged/aliased. Could benefit from a different font.
> * Large font could be a bit bigger to use more of the screen, maybe also other type with anti-aliasing?

Done in last version in development branch.

> * The screen flashes on every update. Using sprites was the solution in another thread, see [here](https://github.com/landonr/lilygo-tdisplays3-esphome/issues/11#issuecomment-1463001112)

Done!

> * Module resets when disconnecting serial, then stops responding and does not show sensor input anymore

I'll investigate this.

> * Network connectivity seems very spotty, at least response to ping requests show more than 50% loss (wifi power save mode?)

I'll investigate this also.

melkati, 2024-01-23:
Never tried the S3 on battery. I will check soon.

Sorry for error. Very last mods were tested only with T-Display (not S3). Please, try again.

hapklaar, 2024-01-23:
> Never tried the S3 on battery. I will check soon.
> 
> Sorry for error. Very last mods were tested only with T-Display (not S3). Please, try again.

Compiles fine now

hapklaar, 2024-01-23:
> > * The small font doesn't look very good on this resolution, jagged/aliased. Could benefit from a different font.
> > * Large font could be a bit bigger to use more of the screen, maybe also other type with anti-aliasing?
> 
> Done in last version in development branch.

New fonts look super!

> > * The screen flashes on every update. Using sprites was the solution in another thread, see [here](https://github.com/landonr/lilygo-tdisplays3-esphome/issues/11#issuecomment-1463001112)
> 
> Done!

Confirmed, fixed!

melkati, 2024-01-23:
Great!

Thank you for testing and reporting back. :-)

hapklaar, 2024-01-26:
@melkati It doesn't with the current settings, but it does when I change
` -DTFT_POWER_ON_BATTERY=38`
to
` -DTFT_POWER_ON_BATTERY=15`
in platformio.ini

hapklaar, 2024-01-26:
Only issue remaining for now is that serial needs to be connected for this device to function.

melkati, 2024-01-26:
Pleaese, @hapklaar Así funcionaba antes, pero con un sitema más rápido para que no se

> @melkati It doesn't with the current settings, but it does when I change ` -DTFT_POWER_ON_BATTERY=38` to ` -DTFT_POWER_ON_BATTERY=15` in platformio.ini

Sorry I wrote the wrong pin.

It's fixed now in last commit to development branch.

melkati, 2024-01-26:
> Only issue remaining for now is that serial needs to be connected for this device to function.

I was able to replicate the issue in a T-Display S3 with SCD41 sensor.
It works fine at my end on a TTGO T-Display with SCD30 sensor.

What sensor are you using?

hapklaar, 2024-01-26:
I'm using an SCD30 on a T-Display-S3. I think it is T-Display-S3 specific, maybe something to do with the CPU speed switching?

melkati, 2024-01-26:
Yes. I've narrowed down the search to the utilityLoop() function. I'm looking at it right now...

melkati, 2024-01-26:
It was not the frequency change what was causing the issue, it was the Serial.flush();
Looks like there is a bug (or inconsistence) with this in the Arduino ESP32 framework.

Can you test with the new code in development branch, please?

hapklaar, 2024-01-26:
Very nice, this works now! Can now test on battery.

Found a new issue:

Voltage reading is a little high with the default reference voltage, 5.7v on USB and 4.5v on full battery. So I adjusted this to 900mv. This causes the screen to auto turn off almost instantly when I disconnect USB. When I raise the reference voltage again, this doesn't happen. 

Reference voltage for esp32 should normally be at or around 1100mv, so maybe the reading on display is just way too high? Together with the curious battery level indicator behavior discussed previously, makes me wonder if there might be an issue with calculation somewhere?

melkati, 2024-01-27:
Thank you for testing and reporting. :-)

The default VRef in CO2 Gadget is 930mV and with this it works very closely to real voltage for all users.

As I told you a few times, if you want the battery indicator to be perfect, you have to spend some time to adjust the three values on the battery configuration menu. VRef should be adjusted so voltage reading (that is shown in the same menu in real time) is close to the real voltaje.

If I remember correctly, turning off the display instantly when disconnecting USB is the expected behavior, if configured to do so.

I will keep an open eye on this. As it's using an external library, I can't do much more right now. I already spent some hours trying to find the fault but didn't.

Going to close this as initial issue resolved. I don't want this to be a catch all issue for any issue with S3. If you have a specific issue with steps to reproduce, open a new issue, please.

commit 3a5fa8ed (2024-01-26): T-Display S3 display on when running on battery. Closes #96
closed 2024-01-27 by melkati