melkati/CO2-Gadget #101: Strange behavior of battery indicator

hapklaar, 2024-01-08:
I have an 18650 connected to the battery connector of the TTGO T-Display and configured the battery to be full at 4.1 volt and empty at 3.0 volt in the menu on the device.

- When I have USB connected, it shows 4.9-5V in blue and a full battery indicator (4 bars).
- Then I disconnect USB and battery indicator still shows 5V in grey, but battery indicator only shows 1 bar
- Next voltage updates to real value of battery output of 3.9V, with only 1 bar
- Eventually it drops to 3.8V and shows 1 or 2 bars intermittently
- when at 3.7V it shows a red battery empty icon with no bars

In other words, the indicator seems to have it's own life, separate from the voltage readout :)

melkati, 2024-01-08:
For a Li Ion battery, full voltage should be 4.2V and empty around 3.5V (3.0V is a dead battery, when at 3.5V the energy remaining in the battery is already very low).

The indicator is slow to respond because the battery measurement is done only every 15 seconds to, in the future, avoid draining the battery with too continuous measurements (but you can change it in variable timeBetweenBatteryRead):

https://github.com/melkati/CO2-Gadget/blob/70aaaa06ea40db3ad947cf2d6dc135d09808b091/CO2_Gadget.ino#L66

Have you adjusted the voltage reference? Maybe it's somewhat off.

If you want the battery indicator to be perfect, you have to spend some time to **adjust the three values on the battery configuration menu**.

Regarding the suggestions for battery indication:

The battery indicador has six levels (not five):

- 4 bars with blue marque (USB connected, or voltage > 4.5V)
- 4 bars with silver marque (Battery level > 80)
- 3 bars with silver marque (Battery level > 60)
- 2 bars with silver marque (Battery level > 40)
- 1 bar with silver marque (Battery level > 20)
- 0 bar in red (Battery level < 20)

https://github.com/melkati/CO2-Gadget/blob/70aaaa06ea40db3ad947cf2d6dc135d09808b091/CO2_Gadget_TFT.h#L133-L158

The battery voltage is measured with an external library.

The battery level is not a percentage, it's the result of a Symmetric Sigmoidal Approximation (to approximate a Li Ion discharge curve, as it's not linear, these batteries are most of the time during discharge at about 3.7-3.8V).

https://github.com/rlogiacco/BatterySense/blob/147417342bff5cc91b3571f5dcfef91b06dc24c1/Battery.h#L97C23-L107

hapklaar, 2024-01-08:
The voltage shown is correct, only the indicator is way off and even changes between three states (2 bars, one bar and 0 bars) on the same voltage. Right now it shows 3.9V and a red battery indicator, this doesn't make any sense.

Would adjusting the voltage reference be able to correct this when the voltage shown is already correct?

melkati, 2024-01-08:
Before writing my previous answer, I connected my CO2 Gadget to a lab power supply and did some testing. Everything looks fine.

I'll reopen this issue and will review and try to reproduce it again.

melkati, 2024-01-10:
Today I spent almost two hours trying to replicate your issue without success.

Maybe it's an issue with your particular hardware?

Please, double or triple check power connections and battery wellness.

I added sending of battery percentage and voltage by MQTT with other measurements (is now in the master branch and will be included on next release). Maybe it helps you logging the data to chase the issue.

I suggest you also spend some time to adjust the three values on the battery configuration menu. It's really important that these values are correct (I have max at 4.2V min at 3.5V and Voltage ref at 930, you must find out the right Voltage ref for your board).

Sincerely, I don't understand what can cause the issue on the firmware side (I don't say it's impossible, just I'm out of ideas). The ADC on the ESP32 is far from perfect and can cause the battery indicator to go up and down by one bar, but I think it can't make  it jump all over the place.

As nobody told me until now about this issue, I'll close the issue until there is new information.

hapklaar, 2024-01-10:
I'll try and debug some more on my side. Also with lab power supply and different batteries. I also have some more T-Display's to test with. Thanks for trying.

hapklaar, 2024-01-13:
Battery indicator still acting strange. Battery sensor went straight from 92 to 14 when voltage was at 3.8v

Plotted behavior in a graph:
[image]

melkati, 2024-01-14:
Thanks for helping me debugging the issue.

I reviewed the code again, line by line (including the external library), and can't find where this issue can come from (if the three settings on the battery configuration menu are correct).

melkati, 2024-02-21:
@hapklaar 

Hi.

I think I fixed the issue with the battery. There were some issues in the code...

Can you compile CO2 Gadget? If so, please test with version in branch [fix_101_battery_voltage](https://github.com/melkati/CO2-Gadget/tree/fix_101_battery_voltage).

If you can't compile, this fix will be on the next release version in a few days after we have the time to do some testing.

hapklaar, 2024-02-21:
I don't use this on battery anymore as the screen fails to stay on on my T-Display-S3 on battery (as specified in https://github.com/melkati/CO2-Gadget/issues/96#issuecomment-1912695396)

It behaves a little different now, as it flashes on and off for a short while after disconnecting USB. When I raise the voltage ref it stays on longer, but readout is too high. When I lower the voltage ref to have a correct voltage readout, screen immediately turns off when disconnecting USB.

Maybe a vid explains better. You can see it starting to go off and on, this will increase in speed as the battery voltage drops a little and eventually after 5-10 minutes the screen will remain off.

https://github.com/melkati/CO2-Gadget/assets/7703320/f2bc0f9b-acde-4c60-8af8-38d24f7cbce9

melkati, 2024-02-22:
Thank you very much, @hapklaar, for recording and uploading the video. Now I have a very clear view of what's happening:

The voltage reference is not right, it's showing 5.3V on USB power and 4.4V on battery. It should show max 4.2V on a fully charged battery and 4.8 to 5.2V on USB (5.3 is out of USB specifications).

CO2 Gadget assumes it is running on external power (probably USB) when the measured voltage is higher than the maximum battery voltage by 5% (4.2V + 5% = ~4.4V).

This is, probably, what is happening in the video:

- 00:00 CO2 Gadget is running on USB power. Voltage = 5.5V so it assumes running on external power. (right)
- 00:08 CO2 Gadget is running on battery. Voltage = 4.4V so it assumes running on external power. (wrong)
- 00:23 Voltage fluctuates just below maximum battery voltage + 5%. CO2 Gadget assumes it's running on battery so turns off the display. (right)
- 00:29 Battery recovers (because now there is no power consumption from the display), so voltage goes up by a little, just below maximum battery voltage + 5%. (wrong)
- 00:43 Cycle repeats... until battery drops by a little because of discharging and now the uncertainty around 4.4V disappears...

How to solve it?

You must make sure (it's really important) that the battery configuration is correct, **you MUST set the maximum battery voltage to 4.2V and minimum battery voltage to 3.5V**. You must have a really very good reason to modify these values (probably I should remove them from the user configuration settings).

It's very important to set the right Voltage reference value (as the resistor tolerances are big, you must find out the right Voltage ref for your board). **The battery voltage for a fully charged li-po or li-ion battery can never be more than 4.2V.**

On my side I already modified the threshold for CO2 Gadget to assume is running on external power from maximum battery voltage + 5% to maximum battery voltage + 9% (~4.58V). It's available at https://github.com/melkati/CO2-Gadget/tree/fix_101_battery_voltage until changes are tested and can merge into the release version.

I hope that the explanation is sufficient to understand what is happening and that the problem is solved with the indicated solutions.

I look forward to hearing from you.

hapklaar, 2024-02-22:
@melkati I think your analysis of the cycle with voltage dropping and rising is correct. 
When I change the voltage reference to 900mv, I have correct voltages, but the display is always off. Your explanation tells me this is by design, the screen should always be off when on battery?

I have some questions which to which the answers still elude me:

1. Why can't I have the display on when on battery on the T-Display-S3 ?
2. What exactly do the maximum and minimum voltage settings do? I thought these values only determined the behaviour of the battery meter bars.

melkati, 2024-02-22:
> @melkati I think your analysis of the cycle with voltage dropping and rising is correct. When I change the voltage reference to 900mv, I have correct voltages, but the display is always off. Your explanation tells me this is by design, the screen should always be off when on battery?
> 

It depends on the Display Config:

[image]

From the blog:

- _Time To Off: Time in seconds without pressing a button for the screen to turn off automatically. Useful for battery saving. The screen will turn on again by briefly pressing the top button. Setting it to 0 disables it (screen always on)._
- _Off on USB: Allows you to select whether you want the screen to turn off when CO2 Gadget is running on external power (USB, for example). In this case it may not be necessary to turn it off to save battery power, but you may want to turn it off to avoid being disturbed by the light if you have it on your bedside table._

https://emariete.com/en/co2-meter-gadget/#Display_Config

> I have some questions which to which the answers still elude me:
> 
> 1. Why can't I have the display on when on battery on the T-Display-S3 ?

The display _should_ remain on when on battery if _Time To Off_ is 0. 

That said, **you are very right**: I connected a battery to my T-Display S3 to understand once and for all what was happening and... the screen turns off when I disconnect the USB or as soon as I exit the menu.

> 2. What exactly do the maximum and minimum voltage settings do? I thought these values only determined the behaviour of the battery meter bars.

These values are used for a few things:

- To know if it's working on battery or on external power, so CO2 Gadget can change the behavior,
- To calculate the battery discharge curve and charge levels.
- To calculate icons and colors for the display

Please, let me look at the issue now that I have a battery connected and, I hope tomorrow or the day after, I will come back to you with the fix.

Please don't install the `fix_101_battery_voltage` branch until I finish with fixing as we are doing testing on that branch and probably it will be in an inconsistent state. Right now, you have the best version installed.

Thank you.

melkati, 2024-02-24:
Hi @hapklaar.

I just uploaded a new version (0.10.13) to **`development`** branch (**not to fix_101_battery_voltage branch** as it maybe unstable) with many fixes related to battery and display on Lilygo T-Display ESP32 S3 board.

I hope everything is fixed now.

Please test it thoroughly and if everything is OK, I will include it in the new release as soon as you confirm it.

Thank you very much for your help.

closed 2024-02-25 by melkati

hapklaar, 2024-02-25:
Testing it on battery now. Module stays on and battery indicator seems to behave much better.

I'll report back with battery voltage and level graph as soon as the battery is empty, which takes longer than I expected. It's already running since 00:18, now at 3.3v and still going strong.

melkati, 2024-02-25:
Thank you @hapklaar 

I just uploaded v0.12.000 as general release with many fixes and improvements.

It's mainly a maintenance release as it doesn't include any new big features. 

Source code is at master, and it's available for easy installing from web browser [here](https://emariete.com/en/co2-meter-gadget/#Instalacion_de_CO2_Gadget_Advanced).

I'm closing the issue for now. Please, feel free it the is still any issue or if you want to give some additional feedback.

Thank you very much for your help reporting and testing all things related to battery and display. You helped make OS2 Gadget a better firmware. :-)

hapklaar, 2024-02-25:
It stopped working reliably below 2.9v. CO2 values were no longer correct. If you check the voltage graph, you could determine battery can be considered empty a bit later than current 3.5v. It still ran for 5 hours after reaching 3.5v until it was at 3.0v. (which is 27% of total runtime)

I would suggest battery empty should be at 3.0 volt.

[image]

Other than this, haven't been able to find any more issues. Well done! I'll keep testing, so might find more later ;)

commit 3a8612ea (2024-02-24): Merge pull request #183 from melkati/fix_101_battery_voltage

Fix #101 battery voltage and Display backlight on Lilygo T-Display ESP32 S3
commit 800949e5 (2024-02-24): Update battery functionality and preferences to fix #101