melkati/CO2-Gadget #111: Some observations after testing current code (issues with MQTT and console)

hapklaar, 2024-01-11:
I compiled and flashed your current code and would like to share my initial observations:

- Upon startup I see all the MQTT topics being published on the console and using MQTT Explorer I can see them appearing. However after this no more messages are received by the MQTT broker, even though console says new values are being published:

```
-->[MQTT] Publishing 701 to topic: homeassistant/co2
-->[MQTT] Publishing 16 to topic: homeassistant/temp
-->[MQTT] Publishing 53 to topic: homeassistant/humi
-->[MQTT] Publishing 752 to topic: homeassistant/uptime
-->[MQTT] Publishing 5 to topic: homeassistant/voltage
-->[MQTT] Publishing 100 to topic: homeassistant/battery
...
```
- It seems no longer possible to navigate the menu via the serial console. Keypresses seem to be ignored. Using the buttons on the esp is still possible, though a bit cumbersome as long press and short press are easily mixen. Maybe make the long press require a longer action.

PS I haven't seen the strange battery indicator anymore, this seems to have resolved itself after removing the device from power and battery for an extended period.

melkati, 2024-01-12:
Thank you for reporting.

I'll try to duplicate the issue with the MQTT messages. At the moment it's working fine at my end.

[image]

I was able to replicate the issue with the serial console.

CO2 2 Gadget was working underclocked to 80Mhz to save battery. In the last few days, I implemented a "dynamic CPU speed". Now if it's working on external power (>4.5V) it works at 240Mhz and if it's working on battery it works at 80Mhz.
The code that makes this change also restarts the serial port speed (as changing CPU speed affects the port speed).
It was resetting the serial port on each loop, making serial input communication impossible. I changed it now so it only happens if a change in speed is made.

https://github.com/melkati/CO2-Gadget/blob/fd84fdd2499596a95b3e781f41b20606d8852072/CO2_Gadget.ino#L374-L381

I hope it's fixed now.

You can modify the long press time by changing LONGCLICK_TIME_MS.

https://github.com/melkati/CO2-Gadget/blob/531adae99ff0df44303aa6c50819b478a576f1aa/CO2_Gadget_Buttons.h#L6C35-L6C86

hapklaar, 2024-01-12:
Console works now. MQTT also works after completely erasing flash and starting fresh. Might have been something left over from previous firmware or something.

ps you should really open a window ;)

hapklaar, 2024-01-12:
I'm playing with MQTT discovery in Home Assistant. Based on https://github.com/melkati/CO2-Gadget/pull/103/commits/b7004174437f9509ccc9b981993d6e2ed99d14a6 I thought this might be working, but can't get the device to be discovered in HA. Is this still WIP?

closed 2024-01-13 by melkati