melkati/CO2-Gadget #116: MQTT discovery in Home Assistant not working?

melkati, 2024-01-13:
I'm playing with MQTT discovery in Home Assistant. Based on https://github.com/melkati/CO2-Gadget/pull/103/commits/b7004174437f9509ccc9b981993d6e2ed99d14a6 I thought this might be working, but can't get the device to be discovered in HA. Is this still WIP?

_Originally posted by @hapklaar in https://github.com/melkati/CO2-Gadget/issues/111#issuecomment-1889096733_

melkati, 2024-01-13:
It's still WIP but it should work.

Can you see the discovery messages sent? It is sent only once when connected to MQTT broker after boot (to resend it you have to reboot CO2 Gadget).

There should be messages sent to topic homeassistant/sensor/SENSOR_NAME/uptime/config

Something similar to this:

[image]

I will not chase this issue as code is changing fast (and will be still changing a lot until the next release).

Please, check your configuration. There aren't too much error checking or sanity checks on this (still developing) so any mistake in configuration will make discovery fail.

Please, tell me if you find out more information.

hapklaar, 2024-01-13:
The config messages weren't there when I checked, there was no entry for it under homeassistant/sensor. 

However, they are present now and device is also auto discovered. Not sure what triggered it, maybe a couple of restarts or just time.

melkati, 2024-01-14:
I will close the issue for now.

closed 2024-01-14 by melkati