melkati/CO2-Gadget #126: Support Mike Ranklin open hardware

samuk, 2024-01-17:
**Is your feature request related to a problem? Please describe.**
It would be great if I could run feature complete CO2 monitor software on open hardware
https://twitter.com/mikerankin/status/1747263018357141833

**Describe the solution you'd like**
Evaluate whether it would run on https://github.com/mike-rankin/ESP32-S3_CO2_Monitor

**Describe alternatives you've considered**
Hacking together closed hardware PCB's

**Additional context**
[image]

melkati, 2024-01-17:
I can't include support for that exact hardware, as I don't have it, but it will be mostly pre-supported very soon.

I have ordered some displays with that chip controller, and also support for ESP32-S3 is planned #96.

closed 2024-01-17 by melkati

melkati, 2024-01-21:
See #96