melkati/CO2-Gadget #132: WiFi Settings over BLE doesn't work if WiFi is dissabled

melkati, 2024-01-22:
@Coscolin:

No es cuestión de reiniciar... Si metes SSID y pass desde el móvil, cuando en el gadget está el WiFi=OFF, no funciona. En realidad no hace nada, porque ni se rellenan los valores de SSID en el gadget, al entrar al menú está en blanco, y al activar el WiFi desde el menú del gadget imagino que intenta conectar al WiFi con los valores que hay en el gadget (que están en blanco).
Una vez activado el WiFi en el gadget, aunque no haya podido conectar porque SSID y pass estaban mal, si configuras esos valores desde el móvil, conecta sin necesidad de reiniciar ni nada.

Lo curioso es que una vez conectado al WiFi, entras al menú del gadget y SSID sigue en blanco, aunque la conexión está bien y activa.

melkati, 2024-01-23:
Checked!

Preferences are not saved.

Also, decide if after setting credentials WiFi must be activated if not active or let the user decide.

melkati, 2024-02-18:
I think the problem was because I missed to activate Wifi after a successful Wifi changed via BLE.

I'll apply the fix in the next release version.

Please, test if it's solved for you. I close this issue for now. I will open it again if you say it's not solved.

Thank you.

closed 2024-02-18 by melkati

Coscolin, 2024-02-22:
I've tested it and it works.

Thank you very much.