melkati/CO2-Gadget #134: Display doesn't redraw completely at exit from menu

melkati, 2024-01-23:
**CO2 Gadget version and flavour**

Paste the two first serial log lines, as:

```
-->[STUP] CO2 Gadget Version: 0.8.046-development Flavour: TTGO T-Display Sandwich
-->[STUP] Version compiled: Jan 23 2024 at 14:10:19
```

**CO2 Gadget installation method?**

Any

**Describe the bug**
Display doesn't redraw completely when exiting from on display menu by timeout (it does by existing manually).
Bug introduced by implementing sprites.

melkati, 2024-01-24:
I think is will be fixed with next update in development branch

melkati, 2024-01-25:
Not completely fixed.

New attempt by including changedToIdle variable to fill screen back when menu timeouts.

Coscolin, 2024-01-26:
I have solved this bug:

https://github.com/melkati/CO2-Gadget/pull/140/commits/b320c3b368e42b80a9a92450c2819cca773f227c

You must add:  out.clear();
in idle function:

#if defined(SUPPORT_TFT) || defined(SUPPORT_OLED)
        out.clear();
        displayShowValues();
#endif

melkati, 2024-01-26:
> I have solved this bug:
> 
> [b320c3b](https://github.com/melkati/CO2-Gadget/commit/b320c3b368e42b80a9a92450c2819cca773f227c)

Good catch!!! Thank you very much, Sergio. 😀

melkati, 2024-01-26:
I had to remove this change as it was causing flickering at each display update. 😒

May be the fix is to do it only when exiting the menu by timeout.

Must investigate how to identify menu exist by timeout to include fix in idle() function.

commit 197f06be (2024-01-24): Clear screen at menu exit. Fixes #134
commit 8a40c5c4 (2024-01-25): Attemp to fix #134
commit f29c7a40 (2024-01-27): Remove battery voltage field from main screen in menu and disable timeout as temporary workaround to fix #134
closed 2024-01-28 by melkati