melkati/CO2-Gadget #135: Backlight turns off in battery when disconnect from USB

melkati, 2024-01-23:
> > * Could you include an option in the display config to turn on the backlight when running on battery? For this module this is off by default and can be enabled by [setting GPIO15 high](https://community.lilygo.cc/topic/711/unable-to-make-t-display-s3-work-with-lipo-battery).
> 
> Already done in the versión you tried. GPIO15 is high all the time.
> 
> There are some issues with display brightness #125. I will come back to this issue in a few days.
> 
> You should configure the "Time to Off" to 0 to disable turning off the display when on battery (this is how it was working before the bug, I hope to fix it).

Just tried with Time to off at 0, but the backlight still turns off when I disconnect USB.

Also wanted to try with a fresh fw, but unfortunately code fails to build with this error:

```
CO2_Gadget_TFT.h: In function 'void setDisplayBrightness(uint32_t)':
CO2_Gadget_TFT.h:158:15: error: 'BACKLIGHT_PWM_CHANNEL' was not declared in this scope
     ledcWrite(BACKLIGHT_PWM_CHANNEL, newBrightness);  // 0-15, 0-255 (with 8 bit resolution); 0=totally dark;255=max brightness
               ^~~~~~~~~~~~~~~~~~~~~
```

_Originally posted by @hapklaar in https://github.com/melkati/CO2-Gadget/issues/96#issuecomment-1905938113_

Coscolin, 2024-01-25:
Creo que este error está relacionado con otro error detectado en el bucle de apagar la pantalla

Con la siguiente configuración, aun conectado por USB se puede reproducir el error para poder hacer DEBUG:
Display Config
[1] Brightness: 10
[2]>Time To Off: 10
[3] Off on USB: ON
[4] Orient: Normal
[5] Temp: Show
[6] Humidity: Show
[7] Battery: Show
[8] <Back-->[MAIN] Turning off display to save power. Actual brightness: 10

-->[TFT ] DisplayBrightness value at 10
-->[TFT ] actualDisplayBrightness value at 10
-->[TFT ] New display brightness value at 0
-->[LOOP] if (actualDisplayBrightness != DisplayBrightness) setDisplayBrightness(DisplayBrightness): 10

-->[TFT ] DisplayBrightness value at 10
-->[TFT ] actualDisplayBrightness value at 0
-->[TFT ] New display brightness value at 10
-->[MAIN] Turning off display to save power. Actual brightness: 10

-->[TFT ] DisplayBrightness value at 10
-->[TFT ] actualDisplayBrightness value at 10
-->[TFT ] New display brightness value at 0
-->[LOOP] if (actualDisplayBrightness != DisplayBrightness) setDisplayBrightness(DisplayBrightness): 10

-->[TFT ] DisplayBrightness value at 10
-->[TFT ] actualDisplayBrightness value at 0
-->[TFT ] New display brightness value at 10
-->[MAIN] Turning off display to save power. Actual brightness: 10

Y si no se cambia termina reiniciándose.

Este es el código a revisar:
void adjustBrightnessLoop() {
#if defined(SUPPORT_OLED) || defined(SUPPORT_TFT)
    if (timeToDisplayOff == 0) return;  // TFT Always ON

    // If battery voltage is more than 5% of the fully charged battery voltage, asume it's working on external power
    boolean workingOnExternalPower = (battery_voltage * 1000 > batteryFullyChargedMillivolts + (batteryFullyChargedMillivolts * 5 / 100));

    // If configured not to turn off the display on external power and it's working on external power, do nothing and return (except if DisplayBrightness is 0))
    if ((!displayOffOnExternalPower) && (workingOnExternalPower)) {
        if (actualDisplayBrightness == 0)  // Exception: When USB connected (just connected) & TFT is OFF -> Turn Display ON
        {
            Serial.println("-->[LOOP] if (actualDisplayBrightness == 0) setDisplayBrightness(DisplayBrightness): " + String(DisplayBrightness));
            setDisplayBrightness(DisplayBrightness);  // Turn on the display
        }
        return;
    }

    if (actualDisplayBrightness != DisplayBrightness) {
        Serial.println("-->[LOOP] if (actualDisplayBrightness != DisplayBrightness) setDisplayBrightness(DisplayBrightness): " + String(DisplayBrightness));
        setDisplayBrightness(DisplayBrightness);
    }

    if ((actualDisplayBrightness != 0) && ( ( millis() - lastTimeButtonPressed ) >= (timeToDisplayOff * 1000))) {
        Serial.println("-->[MAIN] Turning off display to save power. Actual brightness: " + String(actualDisplayBrightness));
        turnOffDisplay();
        actualDisplayBrightness = 0;
    }
#endif
}

Con el [2]>Time To Off: a 900 por ejemplo no pasa.

melkati, 2024-01-27:
This issue is solved in the last development branch. Feedback from user at: https://github.com/melkati/CO2-Gadget/issues/96#issuecomment-1912695396

closed 2024-01-28 by melkati