melkati/CO2-Gadget #151: Sometimes Improv-WiFi do not tigger

melkati, 2024-01-30:
**CO2 Gadget version and flavour**

```
12:17:20.237 > -->[STUP] CO2 Gadget Version: 0.9.001 Flavour: TTGO T-Display Sandwich
12:17:20.237 > -->[STUP] Version compiled: Jan 30 2024 at 11:00:00
```

**CO2 Gadget installation method?**

- Easy Web flasher

**Describe the bug**
Sometimes Improv-WiFi do not tigger and have to go into log console and back into ESP-Web-Tools menu to activate.

Olaruci, 2024-04-20:
Version 0.12.000 Comp Feb 25 2024 TTGO T-Display : VISIT DEVICE & CHANGE WI-FI displayed OK (only when not in menu)
Version 0.12.001 Comp Feb 25 2024 TTGO T-Display : VISIT DEVICE & CHANGE WI-FI displayed OK (only when not in menu)
Version 0.12.022 Comp Apr 10 2024 TTGO T-Display : VISIT DEVICE & CHANGE WI-FI NOT displayed even after multiple retry
Version 0.12.028-new-layout Comp Apr 18 2024 TTGO T-Display : VISIT DEVICE & CHANGE WI-FI NOT displayed even after multiple retry

So sometime between 001 and 022 the functionality as broken but I don't have any intermediate binaries to test.

melkati, 2024-04-24:
Thank you for reporting, Ola.

I'll try to find out what's happening with Improv-WiFi.

melkati, 2024-04-28:
I'm excited to share a potential solution to the Improv-WiFi bug that some of you may have encountered. Before diving into the solution, let's briefly review how Improv-WiFi works and the issue it was causing.

Improv-WiFi operates via the serial port, receiving data from the web application running in your browser (the web installer). When the ESP32 code detects the activation sequence from the web application, a conversation initiates between the browser application and the ESP32 code to perform the necessary tasks.

The problem arose because this clashed with the serial menu system of CO2 Gadget, which continuously reads the serial port. If the serial menu system detects certain characters, it activates the menu. This meant that Improv-WiFi and the serial menu system were competing for control of the serial port, resulting in conflicts where one could potentially "steal" characters intended for the other.

To address this, we've implemented the following solution:

CO2 Gadget now boots with only Improv-WiFi activated, ensuring that Improv-WiFi functions uninterrupted. When the user opens the Improv-WiFi function in the browser, the ESP32's Improv-WiFi exclusively receives characters, preventing any interference.

But how do we ensure that users can still access the serial menu? Here's the trick: before reading each character, the code checks if the next character is "*", indicating the user wants to activate the serial menu. If so, Improv-WiFi is deactivated, and the serial menu takes control of the serial port.

In essence, Improv-WiFi now operates continuously unless the user explicitly activates the serial menu. Once the serial menu is activated, Improv-WiFi is disabled, ensuring smooth functionality for both features.

This solution ensures that Improv-WiFi and the serial menu system can peacefully coexist, providing users with seamless access to both functionalities. Thank you for your patience and understanding as we worked to resolve this issue.

The new beta 0.12.051-development is available both at the development branch in the GitHub repository (https://github.com/melkati/CO2-Gadget/tree/development) and at the Easy Web Installer (https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo).

Just one trick to avoid having to change back and forth between Logs & Console and Main dialogs in your browser to activate Improv:

When you click the blue "Install" button on the web to launch the easy web installer, be sure CO2 Gadget is already initialized so everything in the firmware is already running.

[image]

closed 2024-05-15 by melkati