melkati/CO2-Gadget #155: Temperature never stops increasing/decreasing when temp offset is set

Olaruci, 2024-02-01:
**CO2 Gadget version and flavour**
->[STUP] CO2 Gadget Version: 0.9.001 Flavour: TTGO T-Display
-->[STUP] Version compiled: Jan 30 2024 at 10:59:54

**CO2 Gadget installation method?**
- Easy Web flasher

**Describe the bug**
When a temperature offset is set the displayed temperature never stops incrementing the offset.
So with an offset of -10 it displayed -182 degC and keeps decreasing.

**To Reproduce**
Put any offset from the display menu and it will happen.

**Expected behavior**
Normal offset behavior.

**Screenshots**
[image]

[image]

melkati, 2024-02-02:
Hi. Thanks for reporting this.

What temperature sensor are you using?

melkati, 2024-02-02:
Sensirion sensors calculate offset internally (in their own built-in firmware), CO2 Gadget just pass the offset value to the sensor.

The bug is that the offset must always be a positive number (it will drop temperature by this number and also will recalculate relative humidity).

I'll update the menu to only accept positive numbers.

While I do the modifications, please just change the "-10" temperature offset value to "10").

Note: Be prepared when you set an offset to not see the change immediately. Sensirion's firmware takes a while to adjust it (up to some minutes).

Thank you for reporting it. 😀

Olaruci, 2024-02-03:
[image]
With positive offset it seems like nothing happens. I've let it 24h...

melkati, 2024-02-03:
Thanks for reporting it, I have been able to replicate it and it seems that there are still some errors.

I'm going to look at it in depth because at first glance I can't see anything wrong.

melkati, 2024-02-03:
I think it's fixed now. I tested it and it's working fine on my end.

As the fix is not included in the release version, I compiled a beta version just for you so you can check if it's working fine and report before I insert the modifications in the general release version.

Please, install it from here (be sure to choose the one stated as "CO2 Gadget Beta-Development Version"):
https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo

After flashing, please make sure it's running the right version: CO2 Gadget Version: 0.9.010 Flavour: TTGO T-Display Sandwich

Thank you! :-)

Olaruci, 2024-02-04:
Hi , I don't have the Sandwich vesion, the regular TTGO T-Display

melkati, 2024-02-04:
> Hi, I don't have the Sandwich version, the regular TTGO T-Display

Opps! I'm very sorry, I oversaw that.

I just compiled a new version for you.

Please, install it from here (be sure to choose the one stated as "CO2 Gadget Beta-Development Version"):
https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo

After flashing, please make sure it's running the right version: CO2 Gadget Version: 0.9.014 Flavour: TTGO T-Display

Thank you! 😇

Olaruci, 2024-02-04:
Thanks. 
Yes now the offset is working. And is limited to be positive 0-30.

On information page didn't show 9.014 but is for sure the beta you provided.[image]

[image]

melkati, 2024-02-04:
Great! Thank you.

The fix will be included in the next release in the next few days (we are finishing some more fixes and improvements).

commit 00b1b1ed (2024-02-02): Limit tempOffset to 0-30 degrees on embeded menu (only positive numbers). Fixes #155
closed 2024-02-04 by melkati