melkati/CO2-Gadget #178: Temperature offset only in sensor memory

Olaruci, 2024-02-15:
**CO2 Gadget version and flavour**

Paste the two first serial log lines, as:

```
-->[STUP] CO2 Gadget Version: 0.9.030-development beta: TTGO T-Display 
-->[STUP] Version compiled: Feb 09 2024 
```

**CO2 Gadget installation method?**

- Easy Web flasher? 

**Describe the bug**

If the offset is stored in the sensor (Sensirion SDC30 ) but CO2 Gadget doesn't know it, it can be out of sync and the sensor still adjust the temperature but CO2 Gadget shows the offset at 0.

**To Reproduce**
Steps to reproduce the behavior:
it can happen if, for example, you adjust the offset in CO2 Gadget (and so in the sensor) but later you reset CO2 Gadget preferences by, for example, erasing the flash memory when installing new version. 

**Expected behavior**
Temperature offset inside the sensor and in Co2 Gadget to be in sync.

melkati, 2024-02-18:
Thank you reporting and for your patience,

Now the offset is working right with some fixes we applied internally to the under the hood sensors library https://github.com/kike-canaries/canairio_sensorlib/pull/198 but there is still some issue with temperature offset to be on sync between [sensor <=> library <=> CO2 Gadget].

I'm still working on it. Version 0.9.36 on development already includes some changes but it's not finished.

Sorry for the time it's taking, it's not too easy as it's deeply hidden in the logic and looks like it not only affects the SCD40 sensor but also the SCD4x.

melkati, 2024-02-19:
Hi.

Just to let you know I'm still working on this.

Looks like the bug is deeper than I thought. You can find more information and see advancements here: https://github.com/kike-canaries/canairio_sensorlib/issues/203

Thank you very much for uncovering it.

melkati, 2024-06-07:
I believe this issue has been resolved. Closing now.

closed 2024-06-07 by melkati