melkati/CO2-Gadget #182: When "Time to Off: 0" seconds in battery display off at 0 seconds instead of always on

Coscolin, 2024-02-22:
CO2 Gadget version and flavour:
10:50:12.476 > -->[STUP] CO2 Gadget Version: 0.10.008 Flavour: T-Display S3
10:50:12.476 > -->[STUP] Version compiled: Feb 22 2024 at 10:44:48

CO2 Gadget installation method;
Compiled and uploaded from fix_101_battery_voltage

When "Time to Off: 0" seconds in battery display off at 0 seconds instead of always on

To Reproduce with battery:
1. Set "Time to Off: 0" in display menu
2. Exit menu

To Reproduce with USB:
1. Set "Time to Off: 0" in display menu
2. Set "Off on USB: ON" 
3. Exit menu

Expected behavior
When "Time to Off: 0" display must be always ON.

melkati, 2024-02-24:
Thank you very much for reporting this.

I think this issue is related to #101.

I just uploaded a new version (0.10.13) to `development` branch (**not to fix_101_battery_voltage branch** as it maybe unstable) with many fixes related to battery and display on Lilygo T-Display ESP32 S3 board.

I hope everything is fixed now.

Please test it thoroughly and confirm if everything is fine now.

melkati, 2024-02-25:
Please, see https://github.com/melkati/CO2-Gadget/issues/101#issuecomment-1962961593

Thank you.

closed 2024-02-25 by melkati