melkati/CO2-Gadget #190: Add support for e-Ink (ePaper as Sharewave) displays

melkati, 2024-03-10:


melkati, 2024-03-11:
First basic version uploaded to the [e-Ink CO2 Gadget branch](https://github.com/melkati/CO2-Gadget/tree/e-ink)

Coscolin, 2024-03-12:
As this board has a hardware switch, Do you think is possible to change RST button functionality to use for menu?

Button UP is assigned in platformio to pin 15, but must be to IO39
 -DBTN_UP=39

melkati, 2024-03-12:
> Works on TTGO T5 E-ink GDEM0213B74 with SCD-41 sensor 👏🏻👏🏻👏🏻!!!

Grat! Thank you for testing! 

> Issues detected:

> * a few display refreshments at start, about 12 o 20 times, numbers show very light.

These displays are very tricky to use as they do some flickering and redrawing for some seconds at full redraw.  Also, they tend to dim or get dirty after some partial fast updates without flickering. 

To improve this, after 5 wakeup cycles (it's fixed now but will be configurable by user) the displays do a full refresh to recover quality start fresh again).

I'm still working on improving this. On the displays I have at hand (DEPG0213BN and GDEM029T94) they are already working fairly well. For others we will have to fine tune them.

The problem with these displays is that each one is a different world, that's why it's not easy to support so many of them (especially without having them on hand to test and fine tune).

> * I can not enter in menu with serial or buttons. I suppose that is not implemented yet
> * this board has only two buttons, one for reset and one for up or down, so menu functionality it will be a challenge.

No, it's not implemented. I'm still thinking how this will work.

- Serial menu should be half-working already (and my plan is to fix this completely so serial menu works fine). 
- The e-Ink displays are not designed for fast changes so, probably, an on-display menu will not be doable, in practical terms.
- Because of the lack of buttons and the slowness of e-Ink, probably the only button available on these boards will be for "on demand wake up" of CO2 Gadget. A short press will wakeup it, take a new measurement and show it on the display after a full refresh. A long press will also put CO2 Gadget on an "awake" state for some minutes to give user the opportunity to connect to it via web browser.

> Good job Mario 👏🏻👏🏻👏🏻

Keep into account this is the very first Alpha version (not even beta) with some coding pending to do to be considered beta.

Thank you very much. 😀

Coscolin, 2024-03-15:
### How to implement menu functionality to E-Ink boards with only one button
I would do the following: 

**- In deepSleep:** 
- short press wakes up X seconds, displays measurements and returns to deepSleep 
- long press or double click goes into interactive mode 

**- "awake" in normal display:**
- short press switches between different displays (when available) 
- long press enters menu 
- triple press forces meter into deepSleep 

**- "awake" inside menu:**
- short press changes menu option 
- long press enters into menu option 
- triple click exits menu (back) 
- double click (this option is not so clear to me) changes menu option in reverse of a click 

**- If you are editing a menu option:** 
- one click Up press 
- long press adds up faster and faster 
- 3 seconds without pressing button accepts changes and exits editing

In this way we would have a fully interactive menu with only one button.

melkati, 2024-04-28:
Just a quick update:

There is already a **public beta with support for e-Ink displays** and it's working fairly well.

The new beta 0.12.051-development is available both at the development branch in the GitHub repository (https://github.com/melkati/CO2-Gadget/tree/development) and at the Easy Web Installer (https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo).

This beta adds support for the following e-Ink displays:

- ttgo-t5-EINKBOARDGDEM0213B74: Lilygo TTGO T5 with e-Ink display with panel GDEM0213B74
- ttgo-t5-EINKBOARDDEPG0213BN: Lilygo TTGO T5 with e-Ink display with DEPG0213BN panel
- ttgo-t5-EINKBOARDGDEW0213M21: Lilygo TTGO T5 with e-Ink display with panel GDEW0213M21
- ttgo-t7-EINKBOARDGDEM029T94: Generic ESP32 board with e-Ink display GDEM029T94
- ttgo-t7-WEACT_GDEH0154D67: Generic ESP32 board with WeAct Studio e-Ink display GDEH0154D67
- ttgo-t7-WEACT_DEPG0213BN: Generic ESP32 board with WeAct Studio e-Ink display DEPG0213BN
- ttgo-t7-WEACT_GxEPD2_290_BS: Generic ESP32 board with WeAct Studio e-Ink display GDEM029T94 BS

ttgo-t5-EINKBOARDDEPG0213BN and ttgo-t5-EINKBOARDGDEW0213M21 are working very well, others still need some testing and fine adjustment.

At the moment I recommend the EINKBOARDDEPG0213BN as the one with the best visuals and working better with the current CO2 Gadget version: https://s.click.aliexpress.com/e/_DDFb2gl be sure to choose the board with the DEPG0213BN e-Ink panel).

This board will also be one of the first to support the upcoming LOW POWER version (1-3 months autonomy on battery). More here: https://emariete.com/en/co2-meter-co2-gadget-low-power-with-lilygo-ttgo-t5-epaper-and-sensor-sensirion-scd41/

melkati, 2024-06-03:
This feature has been implemented in the latest release, so I will close the issue for now. Adequate testing for its support in low power mode is pending, but I am optimistic that it will function flawlessly.

closed 2024-06-03 by melkati