melkati/CO2-Gadget #192: Time to off TTGO T-Display

Olaruci, 2024-03-13:
**CO2 Gadget version and flavour**

-->[STUP] CO2 Gadget Version: 0.12.006-beta Flavour: TTGO T-Display
-->[STUP] Version compiled: Feb 29 2024 
```

**CO2 Gadget installation method?**

- Easy Web flasher? 

**Describe the bug**
Can't turn off the Co2-Gadget display. 
Time to off option doesn't turn off the display on battery.
Off on USB : ON doesn't turn off the display on usb power

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Display Config'
2. Select  'Time to OFF.'
3. Wait selected time on battery 
4. See error: display stays  always ON

**Expected behavior**
Display turns off after selected time

![IMG_20240313_155100_897.jpg](https://github.com/melkati/CO2-Gadget/assets/47052441/988e53cf-c022-4475-892d-ddde11556c19)

**Additional context**
I remember this working on older versions

melkati, 2024-03-13:
I was able to reproduce it. Looks like it's affecting all CO2 Gadget flavours.

I will look into it soon.

Thank you for reporting.

melkati, 2024-03-14:
I just uploaded a new beta fixing this issue.

You can install Beta v.0.12.010 from web installer at: https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo

Please, report if it's fixed to close the issue. Thank you.

Olaruci, 2024-03-26:
Hi , sorry for the late replay I just seen this.
But Beta v.0.12.010 doesn't make the screen go off for me.

melkati, 2024-03-27:
Hi Ola.

Please, let me do some more tests with this. 

I just tested it again (flashed a TTGO T-Display Sandwich with firmware Beta v.0.12.010 as yours) and it's working fine connected to USB. Tested with a "Time to of" of 15, 30 and 60 seconds.

@Coscolin, please 🙏, can you test this also to see if it's happening on your side? Beta v.0.12.010 is still available at https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo

Olaruci, 2024-03-28:
I can send a video o telegram maybe I'm doing something wrong,  but I've never seen the screen off with latest firmware on mine.

melkati, 2024-03-29:
Let's try first with a new beta I will upload later today. So we have a clean code base. 

If it's not very complicated for you to configure it again, please install it with the option to clean the memory on install.

I will drop here a message when the new beta is available.

melkati, 2024-03-30:
A new beta (Beta v0.12.016) has been uploaded to:
https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo

**It includes additional debug information about changes on display brightness on the serial log.** You will see lines like:

```
-->[MAIN] Turning off display on external power to save power. Actual brightness: 100
-->[TFT ] DisplayBrightness value at 100
-->[TFT ] actualDisplayBrightness value at 0
-->[TFT ] New display brightness value at 100
```

Maybe this can give us a clue to what is happening.

Just make sure to have a time configure to turn off the display:
[image]

And select to turn off the display also on external power:
[image]

You can set it on the menu or on the preferences.html web page.

Olaruci, 2024-03-31:
No turning off on battery or usb power. And nothing coming on serial, not even the co2,temp,humidity like usual.

Olaruci, 2024-03-31:
I've set it on 10 sec

melkati, 2024-04-01:
> No turning off on battery or usb power. And nothing coming on serial, not even the co2,temp,humidity like usual.

Nothing at all?

Not even something like this at ESP32 start (for example pressing the reset button on the board)?

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (629) esp_core_dump_flash: No core dump �Fѥѥ���found!
E (629) esp_core_dump_flash: No core dump partition found!
E (633) esp_ota_ops: not found otadata
```

melkati, 2024-04-01:
You can also try to reset the ESP32 with the "RESET DEVICE" button on the easy installer web console.

[image]

Olaruci, 2024-04-01:
Yes If I press reset serial information is working: 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv'��0,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (628) esp_core_dump_flash: No core dump p��W�����found!
E (628) esp_core_dump_flash: No core dump partition found!
E (632) esp_ota_ops: not found otadata

-->[STUP] CO2 Gadget Version: 0.12.016 Flavour: TTGO T-Display
-->[STUP] Version compiled: Mar 30 2024 at 06:39:25
-->[STUP] Total heap: 201320
-->[STUP] Free heap: 169804
-->[STUP] Total PSRAM: 0
-->[STUP] Free PSRAM: 0
-->[STUP] Starting up...

-->[IMPR] Init Improv
-->[NPIX] Setting neopixel brightness value at 50
-->[BUZZ] Initializing Buzzer on GPIO 2...
-->[TFT ] Initializing display
4
-->[SBLE] Sensirion Gadget BLE Lib initialized with deviceId = 15:03
-->[SENS] Using sensorlib v0.7.4 Rev:383
-->[SENS] Detecting sensors..
-->[SENS] Selected CO2 Sensor: 0
-->[SENS] Trying to init CO2 sensor: AutoSensor (I2C)
-->[SLIB] temperature offset	: 8.00
-->[SLIB] altitude offset   	: 0.00
-->[SLIB] sea level pressure	: 1013.25 hPa
-->[SLIB] only i2c sensors  	: true
-->[SLIB] sensor registered	: SCD30  	:D
-->[SLIB] sensors count  	: 1 (SCD30,)
-->[SENS]  Sensors detected count	: 1
-->[SENS]  Sensors units count  	: 0
-->[SENS]  Sensors devices names	: SCD30,
-->[SENS] Sensor configured: SCD30

-->[MENU] Use keys + - * /
-->[MENU] to control the menu navigation

-->[STUP] Ready.

I've installed 0.12.019 , I'll test some more tomorrow but for screen I can't see it turning off.

Olaruci, 2024-04-03:
Yeah it's definitely something changed with this turning off function.
I installed 0.11 (downloaded from GitHub and installed with this web tool: https://esp.huhn.me) and the screen is turning off on battery and on USB with no problem.
I could try different version to see when this has been broken if it helps.
But 0.12.019 doesn't help neither.

melkati, 2024-04-05:
I will definitely be interesting if you can try different versions to know in with version it stopped working for you.

v0.12.019 has some additional debug information to try to give a clue on what can be happening.

You should see come entries like:

```
19:31:08.101 > -->[TFT ] DisplayBrightness value at: 100
19:31:08.101 > -->[TFT ] actualDisplayBrightness value at: 0
19:31:08.101 > -->[TFT ] New display brightness value at:-->[SENS] CO2: 968 CO2temp: 20.87 CO2humi: 66.53 H: 0.00 T: 0.00
19:31:08.101 > -->[MAIN] Turning off display on external power to save power. Actual brightness: 100
19:31:08.101 > 
19:31:08.101 > -->[TFT ] DisplayBrightness value at: 100
```

Can you please look at these and paste here some of these blocks (with some lines before and after go give context)?
If you prefer you can download the complete log to a file from the easy web installer and attach it to the reply.

I'm really clueless at what can be happening here. 🤔

Thank you.

Olaruci, 2024-04-06:
So in 0.12 the display is turning off just fine. So from [v0.12.001] onward something broke.
In the latest version after I press reset button and the device restarts if I don't press any button it's turning off the display but after I enter the menu one the turning off doesn't happen anymore. 

Also I don't get those messages in serial only by pressing connect, I have to force them to be displayed by restarting the device.

melkati, 2024-04-07:
> So in 0.12 the display is turning off just fine. So from [v0.12.001] onward something broke.

Please, confirm, in version 0.12.000 it works fine and in version 0.12.001 does not?

> In the latest version after I press reset button and the device restarts if I don't press any button it's turning off the display but after I enter the menu once , the turning off doesn't happen anymore.

While the menu is active it should not turn off the display. 

I guess for you even if you exit the menu, it doesn't turn off anymore?

> Also I don't get those messages in serial only by pressing connect, I have to force them to be displayed by restarting the device.

May be its dependant on the exact OS and drivers you are using. I already saw some differences with different OS/boards/drivers combinations.

If it's working after restarting the board it's fine, it's a minor fault. Anyway, I'll try to find out the reason.

Please, don't forget to put the logs. https://github.com/melkati/CO2-Gadget/issues/192#issuecomment-2040414278

Olaruci, 2024-04-07:
> Please, confirm, in version 0.12.000 it works fine and in version 0.12.001 does not?
Yes
> While the menu is active it should not turn off the display. 
Yes but after a while it get out of the menu by itself anyway if i don't press any button. 
> I guess for you even if you exit the menu, it doesn't turn off anymore?
Yes, exactly. 
> May be its dependant on the exact OS and drivers you are using. I already saw some differences.
 I use linux mint with built-in drivers

> Please, don't forget to put the logs. https://github.com/melkati/CO2-Gadget/issues/192#issuecomment-2040414278
Ok later when I get to the laptop.

Olaruci, 2024-04-07:
also I have to upload only the firmware.bin?
The other like partition file don't change between the versions?

melkati, 2024-04-07:
I think the last change in partitions was between v9 and v10, so if you don't install v9 or older you don't need to write the partitions file (maybe I'm wrong and was between v10 and v11).

Probably you also have to flash spiffs.bin as it contains the web interface as there may be changes or new options or fixes (probably you only have to flash it when upgrading, as from v12.0.000 to v12.0.001 and not when downgrading from v12.0.001 to v12.0.000).

melkati, 2024-04-07:
I just uploaded a new beta reverting the change done between v0.12.000 and v0.12.001

You can install Beta v0.12.021 from the web installer at: https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo

Let's see if it fixes the issue.

Thank you.

Olaruci, 2024-04-08:
Yes I conform that with  v0.12.021 the screen is turning off as expected.
Also the serial messages come immediately as the screen is changing state without having to restart the device:

->[SENS] CO2: 621 CO2temp: 29.20 CO2humi: 29.82 H: 0.00 T: 0.00

-->[TFT ] DisplayBrightness (user setting) value at: 100
-->[TFT ] actualDisplayBrightness value at: 0
-->[TFT ] New display brightness value at: 100
-->[-->[SENS] CO2: 643 CO2temp: 28.62 CO2humi: 31.72 H: 0.00 T: 0.00
-->[MAIN] Turning off display on external power to save power. Actual brightness: 100

-->[TFT ] DisplayBrightness (user setting) value at: 100
-->[TFT ] actualDisplayBrightness value at: 100
-->[TFT ] New display brightness value at: 0
-->[SENS] CO2: 657 CO2temp: 28.12 CO2humi: 31.74 H: 0.00 T: 0.00
-->[MQTT] Publishing 657 to topic: CO2-Gadget/co2
-->[MQTT] Publishing 28 to topic: CO2-Gadget/temp
-->[MQTT] Publishing 32 to topic: CO2-Gadget/humi
-->[MQTT] Publishing 50 to topic: CO2-Gadget/uptime
-->[MQTT] Publishing 5 to topic: CO2-Gadget/voltage
-->[MQTT] Publishing 100 to topic: CO2-Gadget/battery
-->[MQTT] Publishing 59612 to topic: CO2-Gadget/freeMem
-->[MQTT] Publishing -60 to topic: CO2-Gadget/wifiRSSI
-->[MQTT] Publishing 192.168.0.112 to topic: CO2-Gadget/IP
-->[MQTT] Publishing C4:4F:33:6A:15:01 to topic: CO2-Gadget/MAC
-->[MQTT] Publishing CO2-Gadget to topic: CO2-Gadget/hostname
-->[MQTT] Publishing OK to topic: CO2-Gadget/status
-->[SENS] CO2: 652 CO2temp: 27.60 CO2humi: 32.66 H: 0.00 T: 0.00
-->[SENS] CO2: 649 CO2temp: 27.18 CO2humi: 33.51 H: 0.00 T: 0.00

-->[TFT ] DisplayBrightness (user setting) value at: 100
-->[TFT ] actualDisplayBrightness value at: 0
-->[TFT ] New display brightness value at: 100
-->[

CO2 Gadget
[1]>Information
[2] Configuration
[3] Calibration
[4] Reboot w/o saving
[5] <Exit

Information
[1]-Battery 4.99V
[2] Comp Apr  7 2024
[3] Version 0.12.021
[4] TTGO T-Display
[5] Uptime: 0D, 0H, 1M, 17S
[6] IP 192.168.0.112
[7] BLE Dev. Id 15:03                   
[8] <Back

CO2 Gadget
[1]>Information
[2] Configuration
[3] Calibration
[4] Reboot w/o saving
[5] <Exit
-->[MAIN] Turning off display on external power to save power. Actual brightness: 100

-->[TFT ] DisplayBrightness (user setting) value at: 100
-->[TFT ] actualDisplayBrightness value at: 100
-->[TFT ] New display brightness value at: 0
-->[SENS] CO2: 634 CO2temp: 26.51 CO2humi: 35.72 H: 0.00 T: 0.00

-->[TFT ] DisplayBrightness (user setting) value at: 100
-->[TFT ] actualDisplayBrightness value at: 0
-->[TFT ] New display brightness value at: 100
-->[-->[SENS] CO2: 639 CO2temp: 26.12 CO2humi: 35.92 H: 0.00 T: 0.00

melkati, 2024-04-08:
Hi Ola.

I'm glad it's fixed now.

And yes, I also found the reason why serial log was not working.

In fact, the serial console log also stopped working for everybody after entering and exiting the menu: CO2 Gadget stops most logging output while the menu is active (to avoid interrupting it and filling the serial terminal with information while you are inside the menu. A bug caused this "in-menu" detection to malfunction). Surprisingly nobody reported it.

commit d5130686 (2024-03-14): Fix external battery/power detection. Fixes #192
commit 01c9151a (2024-03-14): Beta v0.12.010 Fixes #192
commit 6601b57c (2024-04-07): Removed digitalWrite(TFT_BACKLIGHT, 1) to revert as it was in v.0.12.000 to try to fix #192
commit 6ad67176 (2024-04-08): Commented out extended debug info to fix #192