melkati/CO2-Gadget #197: Blue Visual artefact on screen after wifi disconnected

Olaruci, 2024-03-28:
**CO2 Gadget version and flavour**

-->[STUP] CO2 Gadget Version: 0.12.006 and 0.12.10-beta maybe Flavour: TTGO Display 
```

**CO2 Gadget installation method?**

- Easy Web flasher

**Describe the bug**
After Wi-Fi  or MQTT disconnect some blue highlight lines remains on screen.
After a screen refresh by going in the menu the blue lines are gone.

Might be considered a feature also because you can tell that disconnect happened,  but I'm not sure these are the only situations it happens. :)

**To Reproduce**
Steps to reproduce the behavior:
1. Unplug router or MQTT server for a while.
2. The blue line appears on display if left untouched 

**Expected behavior**
A refresh of whole screen happens after a reconnect. 

**Screenshots**
![20240318_052146.jpg](https://github.com/melkati/CO2-Gadget/assets/47052441/743dce1d-0346-40fd-aa7f-484afedfa6eb)

![20240318_083307.jpg](https://github.com/melkati/CO2-Gadget/assets/47052441/d2121bde-04f0-4f09-aadc-a0242cc8b6a2)

melkati, 2024-03-30:
For the shape and position it looks like the remains of a popup notification.

Let me have some more time to look into it, please.

melkati, 2024-04-01:
I think this issue is fixed.

Let's try first to advance with your other issue (#192) and I will upload a new beta including this fix.

Olaruci, 2024-04-04:
Hi, yes I could not reproduce the issue on 0.12.019 so it looks fixed.
Thanks

melkati, 2024-04-05:
Great!

I will close the issue now. Please, reply if at any time it is malfunctioning again.

closed 2024-04-05 by melkati