melkati/CO2-Gadget #198: **🌟 Calling All Tech Enthusiasts! Join the CO2 Meter Firmware Testing Adventure 🌟**

melkati, 2024-04-08:
**Hello everyone!**

Get ready to be part of something big! I'm looking for enthusiastic volunteers who are interested in collaborating to test and improve the **probably most advanced CO2 monitor firmware in the world, the "CO2 Gadget Advanced Firmware" for ESP32!** This firmware has been in development for four years and is already packed with numerous functionalities. Now, with the latest additions, it's going to be even better! **What's new and exciting for testing is the support for e-ink displays and low-power features.**

Over the past few months, I've poured hundreds of hours into this project, and there comes a point where I need help from others to test it because I simply can't do it all myself. Often, I find myself going in circles, wasting time, because when I make changes or additions, errors can slip in that take a long time to detect by myself. By the time I find them, there have been so many changes since I introduced the error that it becomes very difficult to fix. I end up spending more time fixing these errors than actually moving forward.

That's why I need committed and dedicated individuals. I commit to working with and helping those who participate by addressing the issues they report. But to achieve this, it's crucial to have a small group of people willing to test the new versions on different hardware and conditions.

If you're interested in being part of this testing process, contributing to the firmware's development, and/or working on its optimization and enhancement, please let me know by dropping me a message here. **Your help will be invaluable in making this project a success!**

Just tell me in your message the hardware you have for testing and the kind of help you can bring:

- Test new versions and give feedback (binary versions you can install from the easy web installer)
- Compile from source and test new versions, make small changes to code give feedback 
- Help with development by fixing bugs or developing new additions

You can help with any hardware... **Any CO2 Sensor! Any ESP32 board! Any display (or not display at all)!** All combinations are fine...

Remember: I'm looking for committed and dedicated individuals. I don't want people who will just test the alpha version and forget about it. Together, we can make this firmware more robust and reliable for everyone.

Thank you for considering joining this exciting testing adventure!

josegalvezd, 2024-04-08:
Hi I want to join.
What else i must do?

Regards

Jose

melkati, 2024-04-08:
Hi Jose.

Thank you for joining!

Please, answer this message with this information so I can organize it and know how to follow.

> Just tell me in your message the hardware you have for testing and the kind of help you can bring:
>
> * Test new versions and give feedback (binary versions you can install from the easy web [...]

melkati, 2024-04-09:
> Hi Melkati, I would like to test the new e-ink firmware. I have a llygo T-Display E-Paper 1.02 inch, Lilygo T5 2.13 Inch, and sensirion scd-30
> 
> I can test new versions and give feedback, compile from source and make small changes.

Hi Dominic.

Thank you for joining! 😀

Please, **drop a message at the eMariete's Telegram group https://t.me/emariete_chat** so that I can invite you to the Alpha and Beta Testing Telegram private group.

I haven't tested the SCD30 low power implementation since some time ago (it was working the last time I tried it). I will revisit it soon to make sure at least the basics are working.

Please, tell me the exact display panel that both boards use (there are many different, for example right now DEPG0213BN, GDEM0213B74 and GDEW0213M21 are more or less supported). If possible, attach an image of the inking on the display flex panel connection).

Later we can add together support for the 1.02-inch display, as I don't have one with similar size/resolution to test.

Welcome to the fun! 🤝

melkati, 2024-04-09:
Dominic is already onboard!

Who is next? 😀

Micci253, 2024-04-15:
Hi Melkati! yesterday I received the  llygo T-Display E-Paper 1.02 inch (, Lilygo T5 2.13 Inch (DEPG0213BN) and SCD41 Blue plate .
 (I also have the kit and it works : sensirion scd-30 and LilyGo T-display ) (in Telegram my nickname is Mikhael_N )

melkati, 2024-04-15:
> Hi Mario! yesterday I received the llygo T-Display E-Paper 1.02 inch (, Lilygo T5 2.13 Inch (DEPG0213BN) and SCD41 Blue plate . (I also have the kit and it works : sensirion scd-30 and LilyGo T-display ) (in Telegram my nickname is Mikhael_N )

Hi Mikhael.

Please, read first message:

https://github.com/melkati/CO2-Gadget/issues/198#issue-2230887375

Micci253, 2024-04-16:
Hi Mario! I can :  Compile from source and test new versions, and  make very small changes to code, give feedback. 
I'm not a very good programmer, but I can do something

melkati, 2024-04-16:
> Hi Mario! I can : Compile from source and test new versions, and make very small changes to code, give feedback.
> I'm not a very good programmer, but I can do something

Thank you very much, Mikhael, for joining! 😀

I will send you right now a message via Telegram to invite you to the Alpha and Beta Testing Telegram private group.

The SCD30 low power implementation ir ready for alpha testing and also SCD41 testing will be a very good opportunity to help moving forward.

Welcome to the fun! 🤝

melkati, 2024-04-16:
Mikhael is already onboard!

Who is next? 😀

Olaruci, 2024-04-17:
Hi, 
I can help with testing new versions and give feedback (pre-built binary versions that can install from the easy web installer or other ways )
I have SENSIRION SCD30, SCD41(not tested yet,might be defective) , WINSEN MH-Z14  with LiLYgo T-Display ESP32 BOARD that can be battery powered.

melkati, 2024-04-18:
> Hi, I can help with testing new versions and give feedback (pre-built binary versions that can install from the easy web installer or other ways ) I have SENSIRION SCD30, SCD41(not tested yet,might be defective) , WINSEN MH-Z14 with LiLYgo T-Display ESP32 BOARD that can be battery powered.

Hello Ola!

Thank you so much for offering to help with testing the new versions and providing feedback. Your willingness to contribute is greatly appreciated!

Currently, we don't have pre-built binary versions available, but we're working on it and they'll be ready soon. In the meantime, some other users are compiling their own versions to test.

I've gone ahead and sent you an invitation to join our private Alpha testers group. This way, you'll stay in the loop and be ready to jump in as soon as the binary versions are available.

Thanks again for your willingness to help out! Looking forward to working together on this.

melkati, 2024-04-18:
Ola is already onboard!

Who is next? 😀

melkati, 2024-05-05:
🚀 **Join our Private Alpha Testing Group!** 🚀

The private Alpha testers group is still **open to new members**. Currently, we are a group of five dedicated individuals, but we're looking to expand our team as we have some exciting changes ahead.

With the invaluable help of our current members, we've already implemented support for e-Ink displays, along with numerous improvements and fixes. Now, we're gearing up for **Alpha testing for the real low power functionality**.

If you're passionate about making CO2 Gadget even better and want to contribute, we invite you to join us now!

To participate, all you need is an ESP32 and a CO2 Sensor, along with the ability to compile CO2 Gadget from sources available in the repository here on GitHub.

When reaching out to us, please mention:

1. The hardware you have for testing.
2. The type of help you can provide:
   - Compiling and testing new versions and giving feedback.
   - Compiling from source, testing new versions, making minor code changes, and providing feedback.
   - Contributing to development by fixing bugs or developing new additions (regardless of size, any contribution is valuable).

Your involvement will play a crucial role in shaping the future of CO2 Gadget. We can't wait to welcome you to the team!

Stepyon, 2024-10-10:
Hi,
I would be happy to join the project and help.

**The hardware I have for testing:**
- LOLIN D32 ESP32 board
- SCD41 sensor
- 1.54" 200*200 SSD1680 E-Ink Display ([link](https://www.aliexpress.com/item/1005005196883538.html))
- Also I have a 3D printer and going to design a case to fit the above hardware.

**The type of help I can provide:**
Compiling from source, testing new versions, making code changes, maybe adding new functionality and providing feedback.