melkati/CO2-Gadget #209: Gaps in datapoints sent via BLE to MyAmbiance

melkati, 2024-04-30:
[image] 
https://t.me/c/2135172956/95

melkati, 2024-04-30:
Looks like it was working fine with v0.11.000. https://t.me/c/2135172956/262

[image]

Olaruci, 2024-04-30:
this Screenshot is just the live read.
After I download all it gets messed up.

melkati, 2024-05-01:
> this Screenshot is just the live read. After I download all it gets messed up.

So, you confirm that v0.11.000 did not work well?

This makes it even harder to find the bug because for Dominic (@dominator15) it worked fine with v0.12.007 Flavor TTGO T-Display Sandwich and for me it also worked fine with v0.12.022 Flavor TTGO T-Display Sandwich (the last non beta released and installed from the easy web installer).

melkati, 2024-05-01:
I tested for a couple hours with the last released version on a TTGO T-Display Sandwich (installed from the [Easy Web Installer](https://emariete.com/en/co2-meter-gadget/#Instalacion_de_CO2_Gadget_Advanced)) and after fetching data it looks fine:

```
09:39:46.668 > -->[STUP] CO2 Gadget Version: 0.12.022 Flavour: TTGO T-Display Sandwich
09:39:46.668 > -->[STUP] Version compiled: Apr 10 2024 at 08:02:24
```

[image]

melkati, 2024-05-01:
Shortly after (few minutes) it appeared again (I'm almost sure it appeared later, in live view):

[image]

melkati, 2024-05-02:
Previous test was with libraries:

```
https://github.com/Sensirion/arduino-upt-core.git#b2c0e76
https://github.com/melkati/arduino-ble-gadget.git
```

I'm now compiling the same version to make the same tests again (notice, arduino-upt-core theoretically isn't needed anymore) with libraries:

```
; https://github.com/Sensirion/arduino-upt-core.git
https://github.com/melkati/arduino-ble-gadget.git#historyInterval
```

```
08:44:40.767 > -->[STUP] CO2 Gadget Version: 0.12.022 Flavour: TTGO T-Display Sandwich
08:44:40.767 > -->[STUP] Version compiled: May  2 2024 at 08:43:34
```

I will later update this with the results...

melkati, 2024-05-03:
Looks like there was an issue with library version selection in PlatformIO: As is was installed following Sensirion instructinos as a Global library I think I got mixed results when selecting specific versions in platformio.ini.

In my last tests, looks like it's working fine (using Sensirion's libraries form med April, not installed globally):

```
https://github.com/Sensirion/arduino-upt-core.git#b2c0e76  ; Temporary fix for arduino-ble-gadget incompatibly with arduino-upt-core library
https://github.com/melkati/arduino-ble-gadget.git#98bed70c507bbbaac9a59ebba87f54ae8784c88d

```

[image]

This doesn't work, leaving gaps in data (with last version of Sensirion's libraries):

`    https://github.com/Sensirion/arduino-ble-gadget.git`

melkati, 2024-05-03:
I already reported it to Sensirion https://github.com/Sensirion/arduino-ble-gadget/issues/36

I hope to have an answer soon...

melkati, 2024-05-05:
This issue is already fixed for CO2 Gadget by using libraries:

```
    https://github.com/Sensirion/arduino-upt-core.git#b2c0e76 
    https://github.com/melkati/arduino-ble-gadget.git#historyInterval
```

I'm closing the issue and hope Sensirion fixes it soon (and merges some pull requests) so we can move to use the "official" libraries again.

closed 2024-05-05 by melkati