melkati/CO2-Gadget #216: Issues fetching data from ESP32 into MyAmbiance App in CO2 Gadget Advanced Firmware

melkati, 2024-05-11:
Looks like there is a bug somewhere that is affecting the data fetching from ESP32 into MyAmbiance App.

After nearly 2 weeks almost 100% focused on debugging issues with the combination between ESP32 and MyAmbiance because of this bug and [another one](https://github.com/Sensirion/arduino-ble-gadget/issues/36) (I'm not sure if it's related and is temporarily solved with a workaround already) I'm completely lost at where to look.

I'm still not sure if the issue is related to the CO2 Gadget advanced firmware, the Sensirion ESP32 libs or the MyAmbiance App, I'm completely lost and clueless.

Right now, as the "oficial" Sensirion's library release is not usable for as (because of the data gaps in MyAmbiance) I can't be sure if the issue can be with the combination of libraries CO2 Gadget advanced firmware is using:

```
https://github.com/Sensirion/arduino-upt-core.git#b2c0e76  ; Fix for arduino-ble-gadget incompatibly with arduino-upt-core library
https://github.com/melkati/arduino-ble-gadget.git#historyInterval
```

Right now the issues with Sensirion's library and MyAmbiance are only critical things delaying the release of the new CO2 Gadget advanced firmware version.

The issue:

When fetching data from ESP32 with Sensirion's MyAmbiance the download stops at some point.

- Occasionally, data retrieval is successful, but at other times it fails.
- The issue does not consistently occur at the same point.
- This problem affects all members of the Alpha/Beta testing team.
- Sometimes, retrying results in successful data fetching, while at other times it does not.

[image]

[image]

Olaruci, 2024-05-11:
For me  a clear data & cache of the app after the update to 3.3 resolved the issue.

melkati, 2024-05-12:
I attempted once more (having tried previously) but it continues to fail for me:

[image]

Olaruci, 2024-05-14:
Yes it fails again for me also. The clear cache of app is not a fix.
And yes inconsistent samples numbers also.
[image]

I would like to try previous version 3.2. With that it worked better for me, but I cannot find the apk online (only 2.3 that also doesn't work).
I've let also a comment in play store a comment,  hope they see it.

Coscolin, 2024-05-14:
I have the same issue as @melkati and @Olaruci , even with very few records to download, it fails.

First attempt:
[image]

Second attempt:
[image]

Third atempt:
[image]

Micci253, 2024-05-15:
it fails again.  The clear cache of Sensirion v3.3  application, there were no significant changes. Tried it on devices with Android versions 8,9, Miui 13. 
 I thought that maybe there was a connection with the amount of accumulated data - but even after a small accumulation of data, a reading error may occur. even if before this it was possible to count everything completely in several attempts.
Could the reason be due to some energy-saving features of the bluetooth?

melkati, 2024-05-18:
I compiled the official basic example from Sensirion (Example1) and it's encountering the same issue. To ensure independence from CO2 Gadget Advanced, I installed it from Arduino IDE to isolate it from the CO2 Gadget Advanced firmware developed with VS Code and PlatformIO.

It seems like the issue lies within the library or MyAmbiance (I tested with the new MyAmbiance 3.3.1).

Steps to Reproduce:

Compile and upload the official basic example (Example1) from Sensirion.
Observe the issue persists.
Expected Behavior:
The basic example should function correctly without any data gaps or discrepancies when communicating with the Android phone via MyAmbiance.

Actual Behavior:
The same issue persists, indicating a potential issue within the library or MyAmbiance.

Additional Information:

Tested with MyAmbiance 3.3.1.
Installed from Arduino IDE for isolation from CO2 Gadget Advanced firmware.
This issue seems to occur regardless of the specific project setup, pointing towards a potential issue within the library or MyAmbiance.

[image]

I think it's time to report this to Sensirion.

melkati, 2024-06-03:
Since this is a recognized bug in Sensirion's library, I will close the issue now, with the expectation that the next myAmbiance release will address it. If not, please feel free to reopen the issue. Thank you.

closed 2024-06-03 by melkati