melkati/CO2-Gadget #221: Please add an option to set a static IP address for the CO2-Gadget via the settings.

Airwalk007, 2024-05-18:
One of the best functions is being able to access and use the CO2 gadget via the web and with MQTT. It would be great to be able to set a fixed IP address for the CO2 gadget.

At the moment, the IP address is assigned via DHCP by default. It would be nice to have a choice between DHCP and a fixed IP address.

This option could be included in preferences.html.

melkati, 2024-05-18:
Hi @Airwalk007,

Thank you for your suggestion! It's a great idea, and we're happy to let you know that it's already on our roadmap. You can check out the details here: [[roadmap link].](https://github.com/users/melkati/projects/3/views/1?pane=issue&itemId=52285775)

melkati, 2024-05-20:
Hi @Airwalk007,

I've quickly implemented the option to set a fixed IP address to address your need. Please try it out as soon as possible and provide feedback in this same thread to let us know if it works correctly and meets your expectations.

If there are any errors, it's much easier to fix them sooner rather than later, so your prompt testing and reporting would be greatly appreciated.

The beta version is available for installation via the web at:
[CO2 Gadget Beta Version](https://emariete.com/en/co2-meter-gadget/#CO2_Gadget_Version_Beta-Desarrollo)

Thank you for your suggestion and support!

Airwalk007, 2024-05-20:
Tested with two TTGO_TDISPLAY_SANDWICH and v0.12.76-development ... works great! Thx for your fast implementation.

However, the new function is missing in the version v0.12.77-development ...

melkati, 2024-05-20:
Hi @Airwalk007,

You are right. Somehow the new option disappeared from v0.12.77-development 🤔

I'll look into it right now...

melkati, 2024-05-20:
> Tested with two TTGO_TDISPLAY_SANDWICH and v0.12.76-development ... works great! Thx for your fast implementation.
> 
> However, the new function is missing in the version v0.12.77-development ...

Hi @Airwalk007,

The issue is now fixed (I hope) with v0.12.78-development

Maybe you have to clean your browser cache (or open an inPrivate window).

Sorry for the mistake. 😌

Best regards, 
Mario

closed 2024-05-20 by Airwalk007

melkati, 2024-05-20:
Hi Airwalk007,

I'm thrilled to hear that the latest version, v0.12.78-development, is meeting your expectations! Your feedback was incredibly helpful in ensuring that the fixed IP address feature is working smoothly. It's fantastic to know that this enhancement has made the CO2 gadget even more useful for your daily needs. If you ever come across any more issues or have further suggestions, please feel free to reach out. Your input is invaluable to us.