melkati/CO2-Gadget #224: Reverse orientation doesn't reverse the buttons

Olaruci, 2024-05-20:
**CO2 Gadget version and flavour**

0.12.078

```

**CO2 Gadget installation method?**

- Easy Web flasher? 

**Describe the bug**

 If I do the reverse from the menu the screen and the buttons get reversed as expected.  But if I do it from web the buttons don't even the screen is not reversed immediately,  just after a restart
but the buttons never.
**To Reproduce**
Steps to reproduce the behavior:
Activate reverse from web preferences. 

**Expected behavior**
Screen and buttons get reversed at the same time.

melkati, 2024-05-20:
Thank you very much for reporting this.

In a very first and quick look I noticed also that this functionality is missing for e-Ink displays.

I'll look into it.

melkati, 2024-06-03:
I believe this has been resolved. Could you please confirm so that I can close the issue?

closed 2024-06-07 by melkati